byuflowlab/FLOWUnsteady #1: This is fascinating vortex visulization, but how you made it?

randomwangran, 2020-02-06:
5244dab1d573e4072d9913c304eb953640862afd

It looks like using ParaView.

But how to track the particle and also applied glyphs?

Would you like to share your technique to generate this fascinating video in ParaView?

EdoAlvarezR, 2020-02-11:
The CFD solver is a Lagrangian (meshless) numerical scheme that, instead of having static elements, the computational elements are the particles that convect with the local velocity as shown in the visualization. This is called vortex particle method (VPM).

Check out some of my publications to learn more about the vortex particle method: https://edoalvarezr.github.io/projects/06-publications.html

randomwangran, 2020-02-11:
This is master piece. Really beautiful simulations. 

I've downloaded the work you done during the time at Brigham Young University.

I will play with it and try to, possibily, and find experimental data to support your numerical result.

closed 2020-02-11 by randomwangran