byuflowlab/FLOWUnsteady #25: LoadError (mpmath,scipy) during recompilation

DGCaprace, 2021-01-26:
Starting from a pretty clean Julia install, I get the following error when compiling FLOWUnsteady and all its dependencies:

```
ERROR: LoadError: InitError: PyError ($(Expr(:escape, :(ccall(#= /Users/dg/.julia/packages/PyCall/tqyST/src/pyfncall.jl:43 =# @pysym(:PyObject_Call), PyPtr, (PyPtr, PyPtr, PyPtr), o, pyargsptr, kw))))) <class 'ModuleNotFoundError'>
ModuleNotFoundError("No module named 'mpmath'")
  File "/Users/dg/.julia/conda/3/lib/python3.8/imp.py", line 234, in load_module
    return load_source(name, filename, file)
  File "/Users/dg/.julia/conda/3/lib/python3.8/imp.py", line 171, in load_source
    module = _load(spec)
  File "<frozen importlib._bootstrap>", line 702, in _load
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/Users/dg/.julia/packages/AirfoilPrep/uTtCm/src/airfoilprep.py", line 27, in <module>
    import cmath, mpmath
```

Looks like Conda was correctly installed by Julia, but it misses the required packages `mpmath` and `scipy`.

DGCaprace, 2021-01-26:
I don't know if I missed something in the install or if there is a better way, but I worked around this issue by manually installing the missing packages. In Julia:

```
using PyCall
using Conda
Conda.add("pip")
pip = joinpath(Conda.BINDIR, "pip")
run(`$pip install scipy`)
run(`$pip install mpmath`)
```

Resource: https://discourse.julialang.org/t/manually-installing-python-package-not-found-on-conda/14572

closed 2021-01-26 by DGCaprace

andrewning, 2021-01-27:
I prefer to install my own python rather than rely on Juila's internal version.

EdoAlvarezR, 2021-01-27:
Same over here---I like to use the same python than my system is using for everything else. Checkout this instructions on how to point Julia to the python you want to use: https://flow.byu.edu/FLOWUnsteady/tutorials/installation-instructions/#Setting-up-PyCall