byuflowlab/FLOWUnsteady #37: generate_rotors not found when using examples

ck-bev, 2023-01-17:
when running rotor-related examples (singlerotor.jl, rotornoise/singlerotor.ipynb, rotornoise/singlerotor_BEMT.ipynb), the correct version of generate_rotors is not found. Interestingly, the jupyterlab Contextual Help does find the right method and some other generate_rotors methods are also found. 

```
MethodError: no method matching generate_rotor(::Float64, ::Float64, ::Int64, ::InlineStrings.String31; data_path="/Users/christian/.julia/packages/FLOWUnsteady/UtgRI/src/../data/", pitch=0.0, n=50, CW=true, ReD=940929.447536191, altReD=[5400.0, 0.0001, 1.7308435142352243e-5], verbose=true, xfoil=true, plot_disc=true)
Closest candidates are:
  generate_rotor(::Real, ::Real, ::Int64, ::Matrix{Float64}, ::Matrix{Float64}, ::Matrix{Float64}, ::Matrix{Float64}, ::Vector{Tuple{Float64, Matrix{Float64}, String}}; data_path, read_polar, pitch, n, CW, blade_r, ReD, altReD, Matip, ncrit, xfoil, spline_k, spline_s, splines_s, spline_bc, turbine_flag, rfl_n_lower, rfl_n_upper, rediscretize_airfoils, verbose, verbose_xfoil, v_lvl, save_polars, save_polar_pref, plot_disc, figsize_factor) at ~/.julia/packages/FLOWUnsteady/UtgRI/src/FLOWUnsteady_rotor.jl:30
  generate_rotor(::Real, ::Real, ::Int64, ::Matrix{Float64}, ::Matrix{Float64}, ::Matrix{Float64}, ::Matrix{Float64}, ::Vector{Tuple{Float64, String, String}}; data_path, optargs...) at ~/.julia/packages/FLOWUnsteady/UtgRI/src/FLOWUnsteady_rotor.jl:180
  generate_rotor(::Real, ::Real, ::Int64, ::String; data_path, optargs...) at ~/.julia/packages/FLOWUnsteady/UtgRI/src/FLOWUnsteady_rotor.jl:208

Stacktrace:
 [1] generate_rotor(rotor_file::String; data_path::String, optargs::Base.Pairs{Symbol, Any, NTuple{8, Symbol}, NamedTuple{(:pitch, :n, :CW, :ReD, :altReD, :verbose, :xfoil, :plot_disc), Tuple{Float64, Int64, Bool, Float64, Vector{Float64}, Bool, Bool, Bool}}})
   @ FLOWUnsteady ~/.julia/packages/FLOWUnsteady/UtgRI/src/FLOWUnsteady_rotor.jl:228
 [2] top-level scope
   @ In[12]:61
```

tobias-eh, 2023-01-25:
Maybe this has something to do with the open pull request? It contains a conversion in line 228. 
[see this change](https://github.com/byuflowlab/FLOWUnsteady/pull/36/commits/a1076d4e73bccfef71fdd3101f2ae7a5fc7e70ad)

as a workaround, for me this worked (replacing the InlineString blade_file with a normal String):

`
Rtip, Rhub, B, blade_file = uns.read_rotor(rotor_file; data_path=data_path)
blade_file = "apc10x7_blade.csv"

rotor = uns.generate_rotor(Rtip, Rhub, B, blade_file; data_path=data_path)`

EdoAlvarezR, 2023-04-18:
Solved in [the latest release](https://github.com/byuflowlab/FLOWUnsteady/releases/tag/v3.2.0). Thanks!

closed 2023-04-18 by EdoAlvarezR