byuflowlab/FLOWUnsteady #38: FLOWUnsteady is not detecting FLOWVPM

nanditahari, 2023-02-13:
Using FLOWVPM 3.0 version but its not being detected by FLOWUnsteady. FLOWExaFMM and FLOWVPM passed their tests but there still seems to be some error with compiling FLOWUnsteady. Followed the instructions on https://github.com/byuflowlab/FLOWUnsteady/issues/26 for MACOS but still remains unsolved.

cibinjoseph, 2023-02-24:
I had this same error on a Linux system.
When I run 
```
]add https://github.com/byuflowlab/FLOWUnsteady
```
 Julia reports FLOWVPM is not found even though I've installed all the required dependencies, including `FLOWExaFMM` and `FLOWVPM`.

### Workaround
On inspecting the `Manifest.toml` file in `.julia/environments/v1.8/`, the dependencies of FLOWUnsteady are listed out as:
```
[[deps.FLOWUnsteady]]
deps = ["BPM", "CSV", "DataFrames", "Dates", "Dierckx", "FLOWNoise", 
"FLOWVLM", "GeometricTools", "JLD", "LinearAlgebra", "PyPlot"]
```

`FLOWVPM` is absent from this list of dependencies and on manually adding it, the installation completes succesfully.

### Probable solution
The `Project.toml` file in this repo lists out FLOWVPM as a dependency only in `extras` and not in `deps` ([Ref.](https://github.com/byuflowlab/FLOWUnsteady/blob/401675084cb7caa41268b3d14e3c18c374bf3679/Project.toml#L32))
I think this is the cause of the issue.

EdoAlvarezR, 2023-02-24:
Sorry for the confusion. You guys are on point. FLOWVPM was not open source and publicly shared until I finished my PhD about 6 months ago. For that reason, FLOWUnsteady [checks whether FLOWVPM is available](https://github.com/byuflowlab/FLOWUnsteady/blob/401675084cb7caa41268b3d14e3c18c374bf3679/src/FLOWUnsteady.jl#L27), and if not, it proceeds to generate a [dummy package with the interface of FLOWVPM](https://github.com/byuflowlab/FLOWUnsteady/blob/401675084cb7caa41268b3d14e3c18c374bf3679/src/FLOWUnsteady_dummy_FLOWVPM.jl) for users to still be able to run the quasi-steady solver without the VPM.

Its a very hacky way of declaring a soft dependency. Not very elegant `¯\_(ツ)_/¯`. In order to get it to work, I had to [comment FLOWVPM out](https://github.com/byuflowlab/FLOWUnsteady/blob/401675084cb7caa41268b3d14e3c18c374bf3679/Project.toml#L18) in the Project.toml and just deal with Julia raising a warning that FLOWVPM wasn't declared as a dependency. I wonder if the latest versions of Julia converted that warning into an actual error.

I have fixed that in [v.3.2.0 in the dev branch](https://github.com/byuflowlab/FLOWUnsteady/tree/993524e91ce2e57b838d112ac7e5c3f4c1c4c053). I am currently in the process of rewriting the docs and examples, but feel free to start running your simulations with this newer version. I will be merging it into master and getting it released sometime in the next few days. In the meantime, feel free to peruse through the new documentation running the following in the Julia REPL (make sure you are in the dev branch):

```julia
] add Documenter LiveServer

include("docs/make.jl")

using LiveServer
serve(dir="docs/build")
```

cibinjoseph, 2023-02-25:
I tried installing from the dev branch and it was throwing me a few errors with undefined path variables. `utilities_path` and `examples_path`. I think the best way to guarantee error-free installation would be to test installation on Travis or similar CI systems. Is it the compilation issues with FLOWExaFMM that is keeping you from using CI/CD? I could take a shot at it if you like.

EdoAlvarezR, 2023-02-26:
My bad. I hadn't pushed my changes to FLOWVPM. The dev branch of FLOWVPM is now up to date.

Correct. I haven't been able to automate the compilation of FLOWExaFMM, so the user needs to compile the c++ manually. This poses a roadblock on automating the CI workflow with unit testing. The unit tests are there ([LINK](https://github.com/byuflowlab/FLOWUnsteady/tree/dev/test)), but I can only run them locally (as opposed to letting GitHub automatically run them on every push) =/

Automating compilation of FLOWExaFMM with BinaryBuilder would enable CI/CD, but I have only figured out how to automate Linux (Mac and Windows still missing). The next step would be to register FLOWExaFMM and FLOWVPM in the official Julia registry, then adding those packages would become a simple `] add FLOWExaFMM FLOWVPM` command.

cibinjoseph, 2023-03-28:
The recent updates to the master branch solve this issue. You may close this issue if needed.

EdoAlvarezR, 2023-03-28:
Great! 

I'd recommend checking the new docs if encountering more issues, or for guidance on how to set up simulations: https://flow.byu.edu/FLOWUnsteady
(still working on finishing the last examples and validation cases)