byuflowlab/FLOWUnsteady #49: Issue about wing-example titled 'wing.jl'

inse0918, 2023-04-06:
Hi, I'm new of FLOWUnsteady.

I had built the FLOWUnsteady on my Ubuntu.

The problem is the 'lambda' in GEOMETRY PARAMETERS set at 45.0 deg.

I wanna check the no swept angle (simple rectangular wing). So I changed the 'lambda' from 45.0 to 0.0.

But, the analysis is not functioning normally.

So, I set the 'lambda' 0.1 and 10, but still not working.

Not working means the analysis end even the time step not end.

1. Is there any way to analysis simple rectangular wing with no sweep angle ?

2. How can I check the wake consisted by particle ? I think that the .xmf file contains the data of the particle. But the .xmf file doesn't work in paraview.

Best regards,
Thank you.

Inseo

inse0918, 2023-04-06:
The content of the error is under.

*************************************************************************
START wing-example/wing-example_pfield
        2023-04-06T13:46:16.271
*************************************************************************
        Time step 0 out of 200  Particles: 0
        Time step 10 out of 200 Particles: 1889
        Time step 20 out of 200 Particles: 3899
        Time step 30 out of 200 Particles: 5909
ERROR: LoadError: ExaFMM unexpected error: StackOverflowError()
Stacktrace:
  [1] error(s::String)
    @ Base ./error.jl:35
  [2] call_FLOWExaFMM(pfield::FLOWVPM.ParticleField{Float64, FLOWVPM.ReformulatedVPM{Float64}, FLOWVPM.Inviscid{Float64}, FLOWVPM.NoSFS{Float64}}; verbose::Bool, rbf::Bool, sfs::Bool, sfs_type::Int64, transposed_sfs::Bool, reset::Bool, reset_sfs::Bool, sort::Bool)
    @ FLOWVPM ~/.julia/packages/FLOWVPM/WA9ug/src/FLOWVPM_UJ.jl:164
  [3] call_FLOWExaFMM
    @ ~/.julia/packages/FLOWVPM/WA9ug/src/FLOWVPM_UJ.jl:148 [inlined]
  [4] UJ_fmm(pfield::FLOWVPM.ParticleField{Float64, FLOWVPM.ReformulatedVPM{Float64}, FLOWVPM.Inviscid{Float64}, FLOWVPM.NoSFS{Float64}}; optargs::Base.Pairs{Symbol, Union{}, Tuple{}, NamedTuple{(), Tuple{}}})
    @ FLOWVPM ~/.julia/packages/FLOWVPM/WA9ug/src/FLOWVPM_UJ.jl:119
  [5] UJ_fmm(pfield::FLOWVPM.ParticleField{Float64, FLOWVPM.ReformulatedVPM{Float64}, FLOWVPM.Inviscid{Float64}, FLOWVPM.NoSFS{Float64}})
    @ FLOWVPM ~/.julia/packages/FLOWVPM/WA9ug/src/FLOWVPM_UJ.jl:116
  [6] rungekutta3(pfield::FLOWVPM.ParticleField{Float64, FLOWVPM.ReformulatedVPM{Float64}, FLOWVPM.Inviscid{Float64}, FLOWVPM.NoSFS{Float64}}, dt::Float64; relax::Bool)
    @ FLOWVPM ~/.julia/packages/FLOWVPM/WA9ug/src/FLOWVPM_timeintegration.jl:353
  [7] nextstep(self::FLOWVPM.ParticleField{Float64, FLOWVPM.ReformulatedVPM{Float64}, FLOWVPM.Inviscid{Float64}, FLOWVPM.NoSFS{Float64}}, dt::Float64; optargs::Base.Pairs{Symbol, Bool, Tuple{Symbol}, NamedTuple{(:relax,), Tuple{Bool}}})
    @ FLOWVPM ~/.julia/packages/FLOWVPM/WA9ug/src/FLOWVPM_particlefield.jl:290
  [8] run_vpm!(pfield::FLOWVPM.ParticleField{Float64, FLOWVPM.ReformulatedVPM{Float64}, FLOWVPM.Inviscid{Float64}, FLOWVPM.NoSFS{Float64}}, dt::Float64, nsteps::Int64; runtime_function::FLOWUnsteady.var"#runtime_function#161"{FLOWUnsteady.var"#runtime_function#149#162"{typeof(Vinf), Int64, Float64, Float64, FLOWUnsteady.var"#extra_runtime_function#338"{FLOWUnsteady.var"#extra_runtime_function#329#339"{Vector{Float64}, Vector{Float64}, Bool, FLOWUnsteady.var"#calc_aerodynamicforce#276"{FLOWUnsteady.var"#calc_aerodynamicforce#273#277"{Bool, FLOWUnsteady.var"#calc_aerodynamicforce_kuttajoukowski#285"{FLOWUnsteady.var"#calc_aerodynamicforce_kuttajoukowski#279#286"{Nothing, String, Nothing, Nothing, Bool, Nothing, Nothing, Nothing}}}}, Nothing, Nothing, Nothing, Nothing, String, String, Bool, FLOWUnsteady.var"#336#346", FLOWUnsteady.var"#337#347", Int64, Int64, Bool, FLOWVLM.Wing, typeof(Vinf), Float64, Float64, Float64, Float64, Int64, Float64}}, Int64, Int64, Bool, Tuple{Tuple{Int64, Int64, Float64, Float64}, Tuple{Int64, Int64, Float64, Float64}}, Bool, Bool, Float64, Bool, Bool, Float64, Float64, Vector{Any}, Float64, Float64, Float64, Int64, Float64, String, String, Bool, Int64, Bool, Bool, FLOWUnsteady.Simulation{FLOWUnsteady.UVLMVehicle{0, 0, Float64}, FLOWUnsteady.KinematicManeuver{0, 0}, Float64}, FLOWVPM.ParticleField{Float64, FLOWVPM.ReformulatedVPM{Float64}, FLOWVPM.Inviscid{Float64}, FLOWVPM.NoSFS{Float64}}, Float64}}, static_particles_function::FLOWUnsteady.var"#static_particles_function#10"{Bool, Float64, typeof(FLOWUnsteady.g_pressure), Nothing, String, String, String, Int64, FLOWVPM.ParticleField{Float64, FLOWVPM.ReformulatedVPM{Float64}, FLOWVPM.Inviscid{Float64}, FLOWVPM.NoSFS{Float64}}, FLOWUnsteady.UVLMVehicle{0, 0, Float64}, Float64, Float64, Bool}, save_path::String, create_savepath::Bool, run_name::String, save_code::String, nsteps_save::Int64, prompt::Bool, verbose::Bool, verbose_nsteps::Int64, v_lvl::Int64, save_time::Bool)
    @ FLOWVPM ~/.julia/packages/FLOWVPM/WA9ug/src/FLOWVPM_utils.jl:104
  [9] run_simulation(sim::FLOWUnsteady.Simulation{FLOWUnsteady.UVLMVehicle{0, 0, Float64}, FLOWUnsteady.KinematicManeuver{0, 0}, Float64}, nsteps::Int64; Vinf::typeof(Vinf), sound_spd::Int64, rho::Float64, mu::Float64, tquit::Float64, rand_RPM::Bool, extra_runtime_function::FLOWUnsteady.var"#extra_runtime_function#338"{FLOWUnsteady.var"#extra_runtime_function#329#339"{Vector{Float64}, Vector{Float64}, Bool, FLOWUnsteady.var"#calc_aerodynamicforce#276"{FLOWUnsteady.var"#calc_aerodynamicforce#273#277"{Bool, FLOWUnsteady.var"#calc_aerodynamicforce_kuttajoukowski#285"{FLOWUnsteady.var"#calc_aerodynamicforce_kuttajoukowski#279#286"{Nothing, String, Nothing, Nothing, Bool, Nothing, Nothing, Nothing}}}}, Nothing, Nothing, Nothing, Nothing, String, String, Bool, FLOWUnsteady.var"#336#346", FLOWUnsteady.var"#337#347", Int64, Int64, Bool, FLOWVLM.Wing, typeof(Vinf), Float64, Float64, Float64, Float64, Int64, Float64}}, max_particles::Int64, max_static_particles::Nothing, p_per_step::Int64, vpm_formulation::FLOWVPM.ReformulatedVPM{Float64}, vpm_kernel::FLOWVPM.Kernel, vpm_UJ::typeof(FLOWVPM.UJ_fmm), vpm_SFS::FLOWVPM.NoSFS{Float64}, vpm_integration::typeof(FLOWVPM.rungekutta3), vpm_transposed::Bool, vpm_viscous::FLOWVPM.Inviscid{Float64}, vpm_fmm::FLOWVPM.FMM, vpm_relaxation::FLOWVPM.Relaxation{Float64}, vpm_surface::Bool, vlm_vortexsheet::Bool, vlm_vortexsheet_overlap::Float64, vlm_vortexsheet_d [... cut ...] :Bool, v_lvl::Int64, verbose_nsteps::Int64, raisewarnings::Bool, debug::Bool, nsteps_save::Int64, nsteps_restart::Int64, save_code::String, save_horseshoes::Bool, save_static_particles::Bool, save_wopwopin::Bool)
    @ FLOWUnsteady ~/.julia/packages/FLOWUnsteady/6sUJb/src/FLOWUnsteady_simulation.jl:350
 [10] top-level scope
    @ ~/.julia/packages/FLOWUnsteady/6sUJb/examples/wing/wing.jl:189
 [11] include(fname::String)
    @ Base.MainInclude ./client.jl:476
 [12] top-level scope
    @ REPL[2]:1

EdoAlvarezR, 2023-04-06:
That is the error that ExaFMM throws when a simulation has blown up. If you were to pull up the simulation in ParaView you will see a pretty spectacular explosion of particles haha.

Thanks for pointing this out. I forgot to add relaxation to the VLM solver in that example. I added relaxation in the [latest commits](https://github.com/byuflowlab/FLOWUnsteady/commit/4383ba8fa29a44d1c606ae1c660c3569a55c22cb) and the lambda=0 case now is completely stable.

inse0918, 2023-04-06:
Hi.

Thank you for your hard work.

I checked that the code is going well when the sweep angle is zero.

[image]

There is another question.

Even when I off the paraview( as paraview        = false ), I can earn the files like .xmf , h5 and vtk.

So, the geometry of an wing can be seen when I put the .vtk files into Paraview.

But, there is no particle point that looks like wake just wing geometry.

How can I see the wake consisted of particle ? 

I think that there is pvsm file, but I cannot find that file.

Can't I check the pvsm file if I turn off paraview option in wing.jl file ?

I'm using paraview in Windows 11 because, I tried to build paraview in ubuntu, but there is a obstacle about X11 about displaying monitor(as I know).

Best regards,
Thank you.

Inseo

EdoAlvarezR, 2023-04-06:
To help you (and others) practice visualization, I have now uploaded a practice dataset and added this note at the bottom of [the visualization page ](https://flow.byu.edu/FLOWUnsteady/visualization/) in the docs:

[image]

Datasets:
* High fidelity: [LINK](https://edoalvar2.groups.et.byu.net/public/FLOWUnsteady/rotorhover-highfidelity-particlefdom.zip)
* Mid-high fidelity: [LINK](https://edoalvar2.groups.et.byu.net/public/FLOWUnsteady/rotorhover-midhighfidelity-particlefdom.zip)

Also, did you see this note at the bottom of [the Windows installation instructions](https://flow.byu.edu/FLOWUnsteady/installation/windows/)?
[image]

I hope this helps!

inse0918, 2023-04-07:
Hi.

Thank you, I solved the issue about build paraview in Ubuntu(22.04).  

And I ran the example code and I checked the code is running well.

The problem(?) is that the *.pvsm or the  *.xdmf file is not being created

I wanna check the wake consisted of particles,,,

The example file suggest the *.pvsm file in Mid-high fidelity: [LINK](https://edoalvar2.groups.et.byu.net/public/FLOWUnsteady/rotorhover-midhighfidelity-particlefdom.zip)

Is there any way to create the *.pvsm file ? like as combining the files as *.h5, *.xmf and *.vtk else.

Thank you for your efforts always.

Best regard,
Inseo.

EdoAlvarezR, 2023-04-18:
Apparently you were able to solve this in a different thread: [LINK](https://github.com/byuflowlab/FLOWUnsteady/issues/54#issuecomment-1510606741)

> When I open the *.xmf files in Paraview, the default setting about opening the xdmf files is '**Xdmf3ReaderS**' but when I select the '**XDMF Reade**r', the wake particle can be seen.
> 
> In conclustion
> 
> **Select the XDMF Reader option when opening the post processed files.**
> 
> Thank you. Best regards,
> 
> Inseo

closed 2023-04-18 by EdoAlvarezR