byuflowlab/FLOWUnsteady #53: Issue in compiling FLOWUnsteady

aeroarun, 2023-04-08:
By following the installation instructions, I could successfully install FLOWExaFMM, FLOWVPM and other packages (AirfoilPrep.jl, FLOWVLM, BPM.jl, FLOWNoise). When I finally started installing the FLOWUnsteady package, I ended up with the following error.

[image]

Kindly help me in resolving this issue. Thank you.

EdoAlvarezR, 2023-04-18:
Could you please provide some guidance on how to reproduce this error? It doesn't seem to be an issue [following the instructions](https://flow.byu.edu/FLOWUnsteady/installation/general/) on Linux, MacOs, nor Windows.

MLightD0F9, 2023-04-19:
I have met the same issue during intallation, the Julia version is v1.6.7 LST, the Python version is v3.10.6, which is pre-installed in ubuntu 20.04, the math lib used for python is installed too.
When I install FLOWUnsteady, Firstly, I installed the "GeometricTools.jl", "AirfoilPrep.jl", "FLOWVLM", "BPM.jl", "FLOWNoise", "FLOWPanel.jl" in turn by the way "] add repo". The above all  packages can be installed successfully.
Next, I install the CxxWrap 0.11.2 by the command "import Pkg; Pkg.add(name="CxxWrap", version="0.11.2")", then I install the FLOWExaFMM refer to the old tutorials(add package first, then compile the fmm.so, and at last reprecompile the FLOWExaFMM to finish installation), After FLOWExaFMM is installed, the FLOWVPM is installed by the way "] add https://github.com/byuflowlab/FLOWVPM.jl".
At last, I want to install the FLOWUnsteady by "] add ****", but it can pass through the compile, and when I run "import Pkg; Pkg.precompile()" for more details, I meet the same problem like above.
The same problem also happened for the Julia v1.6.4

EdoAlvarezR, 2023-04-19:
Thanks for confirming that, @MLightD0F9. Could you please upload either a screenshot or copy/paste the stacktrace of the error?

MLightD0F9, 2023-04-19:
I have clone the FLOWUnsteady code in my local machine, and then I add this package, it cannot be compiled. as follow:
[image]
when I precompile for details, it gives me this error:
[image]
the stacktrace of the error is provided as follow:
```
ERROR: The following 1 direct dependency failed to precompile:

FLOWUnsteady [b491798d-ac6e-455f-a27c-49c10bb0a666] 

Failed to precompile FLOWUnsteady [b491798d-ac6e-455f-a27c-49c10bb0a666] to /home/fast/.julia/compiled/v1.6/FLOWUnsteady/jl_puPEzc.
ERROR: LoadError: LoadError: syntax: invalid escape sequence
Stacktrace:
  [1] top-level scope
    @ ~/.julia/packages/FLOWUnsteady/a5QMG/src/FLOWUnsteady_monitors.jl:39
  [2] include(mod::Module, _path::String)
    @ Base ./Base.jl:384
  [3] include
    @ ~/.julia/packages/FLOWUnsteady/a5QMG/src/FLOWUnsteady.jl:11 [inlined]
  [4] top-level scope
    @ ~/.julia/packages/FLOWUnsteady/a5QMG/src/FLOWUnsteady.jl:60
  [5] include
    @ ./Base.jl:384 [inlined]
  [6] include_package_for_output(pkg::Base.PkgId, input::String, depot_path::Vector{String}, dl_load_path::Vector{String}, load_path::Vector{String}, concrete_deps::Vector{Pair{Base.PkgId, UInt64}}, source::Nothing)
    @ Base ./loading.jl:1235
  [7] top-level scope
    @ none:1
  [8] eval
    @ ./boot.jl:360 [inlined]
  [9] eval(x::Expr)
    @ Base.MainInclude ./client.jl:446
 [10] top-level scope
    @ none:1
in expression starting at /home/fast/.julia/packages/FLOWUnsteady/a5QMG/src/FLOWUnsteady_monitors.jl:39
in expression starting at /home/fast/.julia/packages/FLOWUnsteady/a5QMG/src/FLOWUnsteady.jl:1
Stacktrace:
 [1] pkgerror(msg::String)
   @ Pkg.Types /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.6/Pkg/src/Types.jl:55
 [2] precompile(ctx::Pkg.Types.Context; internal_call::Bool, strict::Bool, warn_loaded::Bool, kwargs::Base.Iterators.Pairs{Union{}, Union{}, Tuple{}, NamedTuple{(), Tuple{}}})
   @ Pkg.API /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.6/Pkg/src/API.jl:1265
 [3] precompile
   @ /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.6/Pkg/src/API.jl:921 [inlined]
 [4] #precompile#196
   @ /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.6/Pkg/src/API.jl:919 [inlined]
 [5] precompile()
   @ Pkg.API /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.6/Pkg/src/API.jl:919
 [6] top-level scope
   @ REPL[4]:1
```

EdoAlvarezR, 2023-04-19:
Could you please send a screenshot of what it shows with `] status`? 

Also, it would be helpful if you could attach the following files:
* ~/.julia/environments/v1.6/Project.toml
* ~/.julia/environments/v1.6/Manifest.toml

MLightD0F9, 2023-04-20:
the screenshot after running the `] status` is:
[image]
the Project.toml and Manifest.toml in ~/.julia/environments/v1.6 are provided here, they are packaged and compressed in zip file:
[Project&Manifest.zip](https://github.com/byuflowlab/FLOWUnsteady/files/11279369/Project.Manifest.zip)

MLightD0F9, 2023-04-20:
In my latest test, I upgrade the Julia version from v1.6. to v1.8.5 (only do this change), then the problem is solved, FLOWUnsteady can be compiled successfully, so the reason may be the compatibility between the Julia v1.6 and the Julia v1.8.

aeroarun, 2023-04-20:
By commenting out the triple quoted string literals between the **line numbers 14 to 39 and 230 to 268** in **FLOWUnsteady_monitors.jl** file, I could successfully add the **FLOWUnsteady package** in **Julia 1.6.7** version. Since I am relatively new to Julia language, I could not exactly find out what is the issue in those triple quoted string literals.

EdoAlvarezR, 2023-04-20:
Thanks for dissecting this issue and finding the root cause!

There's now an updated [v3.2.1 release](https://github.com/byuflowlab/FLOWUnsteady/releases/tag/v3.2.1) fixing this issue.

closed 2023-04-20 by EdoAlvarezR