byuflowlab/FLOWUnsteady #54: Question about the results of the blownwing.jl

inse0918, 2023-04-14:
Hi.

I'm following up this program now.

I tested the blownwing.jl code, and I checked the results.

First, when I check the rotation direction of the propellers, all the propeller rotate Inboard-up direction.

It means, the upwash effect can be seen at inboard of the wing section. and the downwash effect can be seen at outboard of the wing section.

But, the section circulation distribution does not.

You can see the opposite results.
(Increased section circulation caused upwash was in outboard of an wing, and decreased section circulation caused downwash was in inboard of an wing.)

[image]

Second, I think that the section circulation distribution is must be proportional to section Cl.

Because, the Lift formulation of VLM is L = rho Vinf Gamma(Circulation).

But the result does not.

[image]

Can you explain this results ?

Thank you.

Best regards,
Inseo.

inse0918, 2023-04-17:
When I open the *.xmf files in Paraview, the default setting about opening the xdmf files is '**Xdmf3ReaderS**' but when I select the '**XDMF Reade**r', the wake particle can be seen.

In conclustion

**Select the XDMF Reader option when opening the post processed files.** 

Thank you.
Best regards,

Inseo

EdoAlvarezR, 2023-04-18:
Thanks for bringing this up Inseo!

The first section of the [Blown Wing tutorial](https://flow.byu.edu/FLOWUnsteady/examples/blownwing-aero) uses the actuator line model (ALM) for the wing because is computationally effective and the intend of the tutorial is to get the user running things quickly and as simple as possible. However, as mentioned in [the second section](https://flow.byu.edu/FLOWUnsteady/examples/blownwing-asm/) of the tutorial, the ALM can lead to unphysical results in cases with strong wake interations (e.g., a rotor wake impinging on a wing surface). For this reason, we have developed and implemented an actuator surface model (ASM) that models the surface more accurately for cases with wake impingement. For more details on the actuator models work, see Chapter 6 and Secs. 8.3 and 8.4 in [my dissertation](https://scholarsarchive.byu.edu/etd/9589/) .

Would you mind running the case with ASM and reporting how the loads compare? Some things to be aware of:

* ASM significantly increases the computational cost (the wing surface uses from 20k to 100k embedded particles to immerse the bound vorticity into the LES)
* The size of the embedded particles in the ASM has a direct effect on the solution as explained here: [LINK](https://flow.byu.edu/FLOWUnsteady/theory/convergence/#Immersed-Vorticity). As much as computational resources permit, we recommend using a particle size in the order of 1/100 of the relevant length scale (in this case, wing thickness).
* It is also possible that the high sweep of the wing  (45 degs) in the tutorial introduces some effects that are counterintuitive and the current prediction (no increase in load in the upwash side) is actually physically accurate. For instance, during validation I came across a similar (but opposite) result on [a blown wing at 10 degs AOA](https://flow.byu.edu/FLOWUnsteady/theory/validation/#Blown-Wing) where there was no drop in load on the downwash side, which turned out to be physically accurate:

[image]

inse0918, 2023-04-19:
Thank you for your kind answer !

I'll apply the ASM code in your [site](https://flow.byu.edu/FLOWUnsteady/examples/blownwing-asm/).

I have a question about Co-axial propeller using FLOWUnsteady.

I made the script about running co-axial propeller.

The code is running well.

[image]

And the problem is the installation angle about the propeller like this figure.

[image]

Is there any way to setting the 'Phase Lag Angle ?

Thank you.

Best regards,
Inseo

inse0918, 2023-04-20:
I found the weird case about the example of the 'blownwing'.

The problem is that the direction of the propellers.

When I ran the 'blownwing.jl' code, the propeller direction is opposite along the free stream wise.

So I change the
<Oaxis = uns.gt.rotation_matrix2(0, 0, 0)            # New orientation>
to 
Oaxis = uns.gt.rotation_matrix2(0, 0, 180)            # New orientation
the propeller direction is going well.

In additionally, if you want to cant or tilt the propeller angle, you can handle the 'Oaxis' variable like I did it. 

Thank you.
Best regards,
Inseo.

EdoAlvarezR, 2023-04-20:
Thanks for catching that! This is just further evidence that visualization is not only for pretty pictures but it is important for verification.

Tutorial and example was corrected in [this commit](https://github.com/byuflowlab/FLOWUnsteady/commit/7d2ae20a33b6104b1442e43b6ac56a3ab3dc04a9). I will update results in the docs later tonight or tomorrow.

closed 2023-04-20 by EdoAlvarezR