byuflowlab/FLOWUnsteady #62: [rotorhover.jl] UndefVarError: control_sigmasensor not defined

inse0918, 2023-04-27:
Hi, Ed.

I changed the vpm_SFS option from
'vpm.SFS_none' to

vpm_SFS       = vpm.DynamicSFS(vpm.Estr_fmm, vpm.pseudo3level_positive;
                                  alpha=0.999, rlxf=0.005, minC=0, maxC=1,
                                  clippings=[vpm.clipping_backscatter],
                                  controls=[control_sigmasensor]
                                  )

But the 'control_sigmasensor' in not defined in this code(rotorhover.jl).

So, the result is,

ERROR: LoadError: UndefVarError: control_sigmasensor not defined 

Could you tell me how to set the 'controls' variable for using the vpm_SFS : vpm.DynamicSFS

Thank you.
Best regards,
Inseo

EdoAlvarezR, 2023-04-27:
Typo fixed in latest dev commit: [LINK](https://github.com/byuflowlab/FLOWUnsteady/commit/1a9f977dc70c4040bc6b135a61c480fdbed43ae8). Thanks for pointing that out!

closed 2023-04-27 by EdoAlvarezR

inse0918, 2023-04-28:
Hi Ed.

I changed the Type to 

vpm_SFS       = vpm.DynamicSFS(vpm.Estr_fmm, vpm.pseudo3level_positive;
                                  alpha=0.999, rlxf=0.005, minC=0, maxC=1,
                                  clippings=[vpm.clipping_backscatter],
                                  controls=[vpm.control_sigmasensor]
                                  )

But the code is still not running with this error ;

ERROR: LoadError: UndefVarError: control_sigmasensor not defined
Stacktrace:
 [1] getproperty(x::Module, f::Symbol)
   @ Base ./Base.jl:31
 [2] top-level scope
   @ ~/.julia/packages/FLOWUnsteady/Jl5A6/examples/rotorhover/rotorhover.jl:178
 [3] include(fname::String)
   @ Base.MainInclude ./client.jl:476
 [4] top-level scope
   @ REPL[1]:1
in expression starting at /home/inse0918/.julia/packages/FLOWUnsteady/Jl5A6/examples/rotorhover/rotorhover.jl:178

! And the other issue !

When I changed the 'rotor_file' from DJI9942.csv to apc10x7

julia> include(joinpath(uns.examples_path, "rotorhover", "rotorhover.jl"))
    RPM:    5400
    Vinf:   [0.0022860000000000003, 0.0, 0.0] m/s
    Matip:  0.21
    ReD:    1.234575e6

Generating geometry...
        Generating geometry...
        Generating airfoil data...
                n4412-1500000.csv
┌ Warning: thread = 1 warning: only found 1 / 4 columns around data row: 1. Filling remaining columns with `missing`
└ @ CSV ~/.julia/packages/CSV/b8ebJ/src/file.jl:576
┌ Warning: thread = 1 warning: only found 2 / 4 columns around data row: 2. Filling remaining columns with `missing`
└ @ CSV ~/.julia/packages/CSV/b8ebJ/src/file.jl:576
┌ Warning: thread = 1 warning: only found 2 / 4 columns around data row: 3. Filling remaining columns with `missing`
└ @ CSV ~/.julia/packages/CSV/b8ebJ/src/file.jl:576
┌ Warning: thread = 1 warning: only found 2 / 4 columns around data row: 4. Filling remaining columns with `missing`
└ @ CSV ~/.julia/packages/CSV/b8ebJ/src/file.jl:576
┌ Warning: thread = 1 warning: only found 2 / 4 columns around data row: 5. Filling remaining columns with `missing`
└ @ CSV ~/.julia/packages/CSV/b8ebJ/src/file.jl:576
┌ Warning: thread = 1 warning: only found 1 / 4 columns around data row: 6. Filling remaining columns with `missing`
└ @ CSV ~/.julia/packages/CSV/b8ebJ/src/file.jl:576
┌ Warning: thread = 1 warning: only found 1 / 4 columns around data row: 7. Filling remaining columns with `missing`
└ @ CSV ~/.julia/packages/CSV/b8ebJ/src/file.jl:576
┌ Warning: thread = 1 warning: parsed expected 4 columns, but didn't reach end of line around data row: 8. Parsing extra columns and widening final columnset
└ @ CSV ~/.julia/packages/CSV/b8ebJ/src/file.jl:577
ERROR: LoadError: MethodError: Cannot `convert` an object of type InlineStrings.String31 to an object of type Float64
Closest candidates are:
  convert(::Type{T}, ::ColorTypes.Gray24) where T<:Real at ~/.julia/packages/ColorTypes/1dGw6/src/conversions.jl:114
  convert(::Type{T}, ::ColorTypes.Gray) where T<:Real at ~/.julia/packages/ColorTypes/1dGw6/src/conversions.jl:113
  convert(::Type{T}, ::Base.TwicePrecision) where T<:Number at twiceprecision.jl:273
  ...
Stacktrace:
  [1] setindex!(A::Vector{Float64}, x::InlineStrings.String31, i1::Int64)
    @ Base ./array.jl:966
  [2] _unsafe_copyto!(dest::Vector{Float64}, doffs::Int64, src::Vector{InlineStrings.String31}, soffs::Int64, n::Int64)
    @ Base ./array.jl:253
  [3] unsafe_copyto!
    @ ./array.jl:307 [inlined]
  [4] _copyto_impl!
    @ ./array.jl:331 [inlined]
  [5] copyto!
[... 8 lines cut ...]
    @ ./array.jl:617 [inlined]
 [10] AirfoilPrep.Polar(init_Re::Int64, init_alpha::Vector{InlineStrings.String31}, init_cl::Vector{Union{Missing, InlineStrings.String31}}, init_cd::Vector{Union{Missing, InlineStrings.String31}}, init_cm::PooledArrays.PooledVector{Union{Missing, InlineStrings.String3}, UInt32, Vector{UInt32}}; Ma::Float64, npanels::Int64, ncrit::Int64, x::Vector{Float64}, y::Vector{Float64}, xsepup::Vector{Float64}, xseplo::Vector{Float64}, pyPolar::PyCall.PyObject)
    @ AirfoilPrep ~/.julia/packages/AirfoilPrep/QE29n/src/AirfoilPrep_pywrapper.jl:54
 [11] read_polar2(file_name::String; path::String, optargs::Base.Pairs{Symbol, Vector{Float64}, Tuple{Symbol, Symbol}, NamedTuple{(:x, :y), Tuple{Vector{Float64}, Vector{Float64}}}})
    @ AirfoilPrep ~/.julia/packages/AirfoilPrep/QE29n/src/AirfoilPrep_pywrapper.jl:120
 [12] generate_rotor(Rtip::Float64, Rhub::Float64, B::Int64, chorddist::Matrix{Float64}, pitchdist::Matrix{Float64}, sweepdist::Matrix{Float64}, heightdist::Matrix{Float64}, airfoil_contours::Vector{Tuple{Float64, Matrix{Float64}, String}}; data_path::String, read_polar::typeof(AirfoilPrep.read_polar2), pitch::Float64, n::Int64, CW::Bool, blade_r::Float64, ReD::Int64, altReD::Vector{Float64}, Matip::Float64, ncrit::Int64, alphas::Vector{Float64}, xfoil::Bool, spline_k::Int64, spline_s::Float64, splines_s::Nothing, spline_bc::String, turbine_flag::Bool, rfl_n_lower::Int64, rfl_n_upper::Int64, rediscretize_airfoils::Bool, verbose::Bool, verbose_xfoil::Bool, v_lvl::Int64, save_polars::Nothing, save_polar_pref::String, plot_disc::Bool, figsize_factor::Float64)
    @ FLOWUnsteady ~/.julia/packages/FLOWUnsteady/Jl5A6/src/FLOWUnsteady_rotor.jl:232
 [13] generate_rotor(Rtip::Float64, Rhub::Float64, B::Int64, chorddist::Matrix{Float64}, pitchdist::Matrix{Float64}, sweepdist::Matrix{Float64}, heightdist::Matrix{Float64}, airfoil_files::Vector{Tuple{Float64, String, String}}; data_path::String, optargs::Base.Pairs{Symbol, Any, NTuple{11 [... cut ...] Prep.read_polar2), Bool, Bool}}})
    @ FLOWUnsteady ~/.julia/packages/FLOWUnsteady/Jl5A6/src/FLOWUnsteady_rotor.jl:355
 [16] top-level scope
    @ ~/.julia/packages/FLOWUnsteady/Jl5A6/examples/rotorhover/rotorhover.jl:256
 [17] include(fname::String)
    @ Base.MainInclude ./client.jl:476
 [18] top-level scope
    @ REPL[1]:1
in expression starting at /home/inse0918/.julia/packages/FLOWUnsteady/Jl5A6/examples/rotorhover/rotorhover.jl:256

How can I fix this issue ?
Thank you
Best regards,
Inseo

filipeinmotion, 2025-03-06:
Did you manage to solve the issue with the csv warning? I am having the same problem