byuflowlab/FLOWUnsteady #69: in vahana.jl

inse0918, 2023-05-25:
Hi, all.

When I changed the vlm_vortexsheet option from 'false' to 'true' to activate ASM.

    *************************************************************************
    START vahana-example/vahana_pfield
            2023-05-25T11:09:03.401
    *************************************************************************
            Time step 0 out of 100  Particles: 0
                    Cd mean, stddev, ratio zeroes:  (NaN, NaN, NaN)
    ERROR: LoadError: ExaFMM unexpected error: StackOverflowError()

The code does not running.

How can I run this code with using ASM ?

Actually, I wanna write the Propeller-wing Interaction script by using ASM.

So, I refer the blownwing.jl applying the ASM contents, but the code is not running...

Anybody got script about the propeller-wing interaction using ASM ?

Thank you
Best regards,
Inseo

closed 2023-06-07 by inse0918

inse0918, 2023-06-08:
Hi, @EdoAlvarezR 

I could not found the exact cause about this error.
Actually, the reason why I posted the question about the ASM is to analysis the propeller-wing interaction.
Anyway, I resolved the problem using original vahana.jl code.

I changed the vahana.jl to PROWIM validation code.
But the result is not similar with your validation data
[image]

My data is,
[image]

The section cl in the wing root is very low.

If you don't mind, could you please check my PROWIM code ?

or, could you please share your PROWIM validation data or code ?

Thank you.
Best regards,
Inseo.

[vahana.zip](https://github.com/byuflowlab/FLOWUnsteady/files/11682770/vahana.zip)

inse0918, 2023-06-08:
Hi, @tim-chron .

I'm glad that you resolved the stackoverflow problem.

I also have the script that analysis the multi-rotor or propeller as you said.

You can handle the number of the propeller which have specific position, angle and phase angle.

Here is my script.

I hope you earn data what you want.

Thank you.
Best regards,
Inseo.

[PropMaster.zip](https://github.com/byuflowlab/FLOWUnsteady/files/11682812/PropMaster.zip)

EvenW254, 2025-05-19:
> [@inse0918](https://github.com/inse0918), any chance you could share how you resolved the problem?

It may be "WAKE TREATMENT".
I meet the same question, I solved it by changing "minmaxGamma".