byuflowlab/FLOWUnsteady #74: Adding "other Packages" and "Testing FlowUnsteady" Fails

philippmandl, 2023-06-19:
Dear All, 

after days of intallation of FLOWUnsteady I am currently figuring out the following issue (after running successfully the pre-defined test procedurces of FLOWExaFMM and FLOWVPM):

julia> Pkg.add([ Pkg.PackageSpec(; url=url*name, rev=v) for (name, v) in packages ])
   Resolving package versions...
  Downloaded artifact: xfoil_light
    Updating `~/.julia/environments/v1.8/Project.toml`
  [008c61af] + AirfoilPrep v2.1.2 `https://github.com/byuflowlab/AirfoilPrep.jl#v2.1.2`
  [f91b385c] + BPM v2.0.1 `https://github.com/byuflowlab/BPM.jl#v2.0.1`
  [d27480ee] + FLOWNoise v2.3.1 `https://github.com/byuflowlab/FLOWNoise#v2.3.1`
  [1a3ff0be] + FLOWVLM v2.1.2 `https://github.com/byuflowlab/FLOWVLM#v2.1.2`
    Updating `~/.julia/environments/v1.8/Manifest.toml`
  [621f4979] + AbstractFFTs v1.3.1
  [79e6a3ab] + Adapt v3.6.2
  [008c61af] + AirfoilPrep v2.1.2 `https://github.com/byuflowlab/AirfoilPrep.jl#v2.1.2`
  [13072b0f] + AxisAlgorithms v1.0.1
  [f91b385c] + BPM v2.0.1 `https://github.com/byuflowlab/BPM.jl#v2.0.1`
⌃ [e1828068] + CCBlade v0.1.0
  [336ed68f] + CSV v0.10.11
  [944b1d66] + CodecZlib v0.7.1
  [a8cc5b0e] + Crayons v4.1.1
  [9a962f9c] + DataAPI v1.15.0
  [a93c6f00] + DataFrames v1.5.0
  [e2d170a0] + DataValueInterfaces v1.0.0
  [7a1cc6ca] + FFTW v1.7.0
  [6cb5d3fb] + FLOWMath v0.3.3
  [d27480ee] + FLOWNoise v2.3.1 `https://github.com/byuflowlab/FLOWNoise#v2.3.1`
  [1a3ff0be] + FLOWVLM v2.1.2 `https://github.com/byuflowlab/FLOWVLM#v2.1.2`
  [48062228] + FilePathsBase v0.9.20
  [59287772] + Formatting v0.4.2
  [842dd82b] + InlineStrings v1.4.0
  [a98d9a8b] + Interpolations v0.14.7
  [41ab1584] + InvertedIndices v1.3.0
  [82899510] + IteratorInterfaceExtensions v1.0.0
  [e1d29d7a] + Missings v1.1.0
  [6fe1bfb0] + OffsetArrays v1.12.9
  [2dfb63ee] + PooledArrays v1.4.2
  [08abe8d2] + PrettyTables v2.2.4
  [c84ed2f1] + Ratios v0.4.5
  [91c51154] + SentinelArrays v1.4.0
  [66db9d55] + SnoopPrecompile v1.0.3
  [a2af1166] + SortingAlgorithms v1.1.1
  [90137ffa] + StaticArrays v1.5.26
  [892a3eda] + StringManipulation v0.3.0
  [3783bdb8] + TableTraits v1.0.1
  [bd369af6] + Tables v1.10.1
  [3bb67fe8] + TranscodingStreams v0.9.13
  [8149f6b0] + WAV v1.2.0
  [ea10d353] + WeakRefStrings v1.4.2
  [efce3f68] + WoodburyMatrices v0.5.5
  [76eceee3] + WorkerUtilities v1.6.1
  [19641d66] + Xfoil v0.5.0
  [f5851436] + FFTW_jll v3.3.10+0
  [1d5cc7b8] + IntelOpenMP_jll v2023.1.0+0
⌅ [856f044c] + MKL_jll v2022.2.0+0
  [70cc596b] + xfoil_light_jll v0.2.1+0
  [8ba89e20] + Distributed
  [1a1011a3] + SharedArrays
        Info Packages marked with ⌃ and ⌅ have new versions available, but those with ⌅ are restricted by compatibility constraints from upgrading. To see why use `status --outdated -m`
Precompiling project...
  ✗ FLOWVLM
  32 dependencies successfully precompiled in 407 seconds. 81 already precompiled.
  1 dependency errored. To see a full report either run `import Pkg; Pkg.precompile()` or load the package

(@v1.8) pkg> add https://github.com/byuflowlab/FLOWUnsteady
    Updating git-repo `https://github.com/byuflowlab/FLOWUnsteady`
   Resolving package versions...
    Updating `~/.julia/environments/v1.8/Project.toml`
  [b491798d] + FLOWUnsteady v3.2.1 `https://github.com/byuflowlab/FLOWUnsteady#master`
[... 106 lines cut ...]
 [5] _require_prelocked(uuidkey::Base.PkgId)
   @ Base ./loading.jl:1200
 [6] macro expansion
   @ ./loading.jl:1180 [inlined]
 [7] macro expansion
   @ ./lock.jl:223 [inlined]
 [8] require(into::Module, mod::Symbol)
   @ Base ./loading.jl:1144

(@v1.8) pkg> status
Status `~/.julia/environments/v1.8/Project.toml`
  [008c61af] AirfoilPrep v2.1.2 `https://github.com/byuflowlab/AirfoilPrep.jl#v2.1.2`
  [f91b385c] BPM v2.0.1 `https://github.com/byuflowlab/BPM.jl#v2.0.1`
⌃ [1f15a43c] CxxWrap v0.11.2
  [a07d1f4e] FLOWExaFMM v2.1.0 `~/Software/FLOWExaFMM`
  [d27480ee] FLOWNoise v2.3.1 `https://github.com/byuflowlab/FLOWNoise#v2.3.1`
  [b491798d] FLOWUnsteady v3.2.1 `https://github.com/byuflowlab/FLOWUnsteady#master`
  [1a3ff0be] FLOWVLM v2.1.2 `https://github.com/byuflowlab/FLOWVLM#v2.1.2`
  [6e19019d] FLOWVPM v3.0.1 `https://github.com/byuflowlab/FLOWVPM.jl#master`
  [83792f5e] GeometricTools v2.1.6
  [7073ff75] IJulia v1.24.2
  [438e738f] PyCall v1.95.2
Info Packages marked with ⌃ have new versions available and may be upgradable.

Do you have any idea what might be wrong? Thank you so much for any help!!

EdoAlvarezR, 2023-06-19:
@philippmandl, could you please comment on how you were able to resolve the issue?

philippmandl, 2023-06-20:
> there was a bug in the python-path to anaconda 3 which wasn't found by the system.
> 
> An additional failure comes up when I try to install the software on Scientific Linux 7.9:
> 
>     * the error message is linked to the ] test FLOWExaFMM execution-
> 
> 
> **Error-Message:**
> 
> pkg> test FLOWExaFMM Testing FLOWExaFMM Status `/tmp/jl_aVi9OH/Project.toml` [1f15a43c] CxxWrap v0.11.2 [a07d1f4e] FLOWExaFMM v2.1.0 `/local/disk1/pmandl/01_Software/FLOWExaFMM` [8dfed614] Test `@stdlib/Test` Status `/tmp/jl_aVi9OH/Manifest.toml` [1f15a43c] CxxWrap v0.11.2 [a07d1f4e] FLOWExaFMM v2.1.0 `/local/disk1/pmandl/01_Software/FLOWExaFMM` [692b3bcd] JLLWrappers v1.4.1 [1914dd2f] MacroTools v0.5.10 [21216c6a] Preferences v1.4.0 [3eaa8342] libcxxwrap_julia_jll v0.8.8+1 [0dad84c5] ArgTools `@stdlib/ArgTools` [56f22d72] Artifacts `@stdlib/Artifacts` [2a0f44e3] Base64 `@stdlib/Base64` [ade2ca70] Dates `@stdlib/Dates` [f43a241f] Downloads `@stdlib/Downloads` [b77e0a4c] InteractiveUtils `@stdlib/InteractiveUtils` [b27032c2] LibCURL `@stdlib/LibCURL` [76f85450] LibGit2 `@stdlib/LibGit2` [8f399da3] Libdl `@stdlib/Libdl` [56ddb016] Logging `@stdlib/Logging` [d6f4376e] Markdown `@stdlib/Markdown` [ca575930] NetworkOptions `@stdlib/NetworkOptions` [44cfe95a] Pkg `@stdlib/Pkg` [de0858da] Printf `@stdlib/Printf` [3fa0cd96] REPL `@stdlib/REPL` [9a3f8284] Random `@stdlib/Random` [ea8e919c] SHA `@stdlib/SHA` [9e88b42a] Serialization `@stdlib/Serialization` [6462fe0b] Sockets `@stdlib/Sockets` [fa267f1f] TOML `@stdlib/TOML` [a4e569a6] Tar `@stdlib/Tar` [8dfed614] Test `@stdlib/Test` [cf7118a7] UUIDs `@stdlib/UUIDs` [4ec0a83e] Unicode `@stdlib/Unicode` [deac9b47] LibCURL_jll `@stdlib/LibCURL_jll` [29816b5a] LibSSH2_jll `@stdlib/LibSSH2_jll` [c8ffd9c3] MbedTLS_jll `@stdlib/MbedTLS_jll` [14a3606d] MozillaCACerts_jll `@stdlib/MozillaCACerts_jll` [83775a58] Zlib_jll `@stdlib/Zlib_jll` [8e850ede] nghttp2_jll `@stdlib/nghttp2_jll` [3f19e933] p7zip_jll `@stdlib/p7zip_jll` Precompiling project... ✗ FLOWExaFMM 0 dependencies successfully precompiled in 1 seconds (8 already precompiled) 1 dependency errored. To see a full report either run `import Pkg; Pkg.precompile()` or load the package Testing Running tests... /local/disk1/pmandl/01_Software/julia-1.6.7/bin/julia: symbol lookup error: /local/disk1/pmandl/01_Software/FLOWExaFMM/src/fmm.so: undefined symbol: _ZN5jlcxx6Module12get_constantERKSs ERROR: LoadError: Failed to precompile FLOWExaFMM [a07d1f4e-0e34-4d8b-bfef-e5b961477d34] to /home/pmandl/.julia/compiled/v1.6/FLOWExaFMM/jl_T5D5wX. Stacktrace: [1] error(s::String) @ Base ./error.jl:33 [2] compilecache(pkg::Base.PkgId, path::String, internal_stderr::Base.TTY, internal_stdout::Base.TTY, ignore_loaded_modules::Bool) @ Base ./loading.jl:1385 [3] compilecache(pkg::Base.PkgId, path::String) @ Base ./loading.jl:1329 [4] _require(pkg::Base.PkgId) @ Base ./loading.jl:1043 [5] require(uuidkey::Base.PkgId) @ Base ./loading.jl:936 [6] require(into::Module, mod::Symbol) @ Base ./loading.jl:923 [7] include(fname::String) @ Base.MainInclude ./client.jl:444 [8] top-level scope @ none:6 in expression starting at /local/disk1/pmandl/01_Software/FLOWExaFMM/test/runtests.jl:2 ERROR: Package FLOWExaFMM errored during testing
> 
> _do you have any ideas which might help? Thank you very much for your support!_

EdoAlvarezR, 2023-06-20:
Are you following the instructions in the docs? [https://flow.byu.edu/FLOWUnsteady/installation/general/#FLOWVPM](https://flow.byu.edu/FLOWUnsteady/installation/general/#FLOWVPM)

As indicated in the docs:
> Test that you can correctly compile C++ code wrapped for Julia following these instructions: [LINK](https://nbviewer.org/github/byuflowlab/FLOWVPM.jl/blob/master/docs/installation-linux.ipynb)

philippmandl, 2023-06-20:
yes, I followed both instructions and the C++ wrapper works correctly - all the tests are passed. I have tried it with Juli 1.6.7, 1.8.5 and 1.9.1 - but always the same problem at the scientific linux system - do you have any ideas?

EdoAlvarezR, 2023-06-21:
Can you please confirm that the file `/local/disk1/pmandl/01_Software/FLOWExaFMM/src/fmm.so` exists? If not, the compilation didn't work.

philippmandl, 2023-06-21:
Yes, this file does exist - that's why I am wondering, why the Test fails.

philippmandl, 2023-06-21:
when running the precompiler, following error is still present 

julia> import Pkg; Pkg.precompile()
Precompiling project...
  ✗ FLOWExaFMM
  ✗ FLOWVPM
  66 dependencies successfully precompiled in 29 seconds

ERROR: The following 2 direct dependencies failed to precompile:

FLOWExaFMM [a07d1f4e-0e34-4d8b-bfef-e5b961477d34]

Failed to precompile FLOWExaFMM [a07d1f4e-0e34-4d8b-bfef-e5b961477d34] to /home/pmandl/.julia/compiled/v1.6/FLOWExaFMM/jl_Nn0DUu.
/local/disk1/pmandl/01_Software/julia-1.6.7/bin/julia: symbol lookup error: /local/disk1/pmandl/01_Software/FLOWExaFMM/src/fmm.so: undefined symbol: _ZN5jlcxx6Module12get_constantERKSs

FLOWVPM [6e19019d-7c31-4940-9d16-c3f15dfe6020]

Failed to precompile FLOWVPM [6e19019d-7c31-4940-9d16-c3f15dfe6020] to /home/pmandl/.julia/compiled/v1.6/FLOWVPM/jl_EC2Pyj.
/local/disk1/pmandl/01_Software/julia-1.6.7/bin/julia: symbol lookup error: /local/disk1/pmandl/01_Software/FLOWExaFMM/src/fmm.so: undefined symbol: _ZN5jlcxx6Module12get_constantERKSs
ERROR: LoadError: Failed to precompile FLOWExaFMM [a07d1f4e-0e34-4d8b-bfef-e5b961477d34] to /home/pmandl/.julia/compiled/v1.6/FLOWExaFMM/jl_8ibDQS.
Stacktrace:
  [1] error(s::String)
    @ Base ./error.jl:33
  [2] compilecache(pkg::Base.PkgId, path::String, internal_stderr::IOContext{Base.PipeEndpoint}, internal_stdout::IOContext{IOStream}, ignore_loaded_modules::Bool)
    @ Base ./loading.jl:1385
  [3] compilecache(pkg::Base.PkgId, path::String)
    @ Base ./loading.jl:1329
  [4] _require(pkg::Base.PkgId)
    @ Base ./loading.jl:1043
  [5] require(uuidkey::Base.PkgId)
    @ Base ./loading.jl:936
  [6] require(into::Module, mod::Symbol)
    @ Base ./loading.jl:923
  [7] include
    @ ./Base.jl:384 [inlined]
  [8] include_package_for_output(pkg::Base.PkgId, input::String, depot_path::Vector{String}, dl_load_path::Vector{String}, load_path::Vector{String}, concrete_deps::Vector{Pair{Base.PkgId, UInt64}}, source::Nothing)
    @ Base ./loading.jl:1235
  [9] top-level scope
    @ none:1
 [10] eval
    @ ./boot.jl:360 [inlined]
 [11] eval(x::Expr)
    @ Base.MainInclude ./client.jl:446
 [12] top-level scope
    @ none:1
in expression starting at /home/pmandl/.julia/packages/FLOWVPM/WA9ug/src/FLOWVPM.jl:1

Stacktrace:
 [1] pkgerror(msg::String)
   @ Pkg.Types /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.6/Pkg/src/Types.jl:55
 [2] precompile(ctx::Pkg.Types.Context; internal_call::Bool, strict::Bool, warn_loaded::Bool, kwargs::Base.Iterators.Pairs{Union{}, Union{}, Tuple{}, NamedTuple{(), Tuple{}}})
   @ Pkg.API /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.6/Pkg/src/API.jl:1265
 [3] precompile
   @ /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.6/Pkg/src/API.jl:921 [inlined]
 [4] #precompile#196
   @ /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.6/Pkg/src/API.jl:919 [inlined]
 [5] precompile()
   @ Pkg.API /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.6/Pkg/src/API.jl:919
 [6] top-level scope
   @ REPL[7]:1

do you have any ideas regarding this problem? I am running out of ideas currently

philippmandl, 2023-06-21:
running test FLOWExaFMM leads to the error

Testing FLOWExaFMM
      Status `/tmp/jl_MB0FLT/Project.toml`
  [1f15a43c] CxxWrap v0.11.2
  [a07d1f4e] FLOWExaFMM v2.1.0 `/local/disk1/pmandl/01_Software/FLOWExaFMM`
  [8dfed614] Test `@stdlib/Test`
      Status `/tmp/jl_MB0FLT/Manifest.toml`
  [1f15a43c] CxxWrap v0.11.2
  [a07d1f4e] FLOWExaFMM v2.1.0 `/local/disk1/pmandl/01_Software/FLOWExaFMM`
  [692b3bcd] JLLWrappers v1.4.1
  [1914dd2f] MacroTools v0.5.10
  [21216c6a] Preferences v1.4.0
  [3eaa8342] libcxxwrap_julia_jll v0.8.8+1
  [0dad84c5] ArgTools `@stdlib/ArgTools`
  [56f22d72] Artifacts `@stdlib/Artifacts`
  [2a0f44e3] Base64 `@stdlib/Base64`
  [ade2ca70] Dates `@stdlib/Dates`
  [f43a241f] Downloads `@stdlib/Downloads`
  [b77e0a4c] InteractiveUtils `@stdlib/InteractiveUtils`
  [b27032c2] LibCURL `@stdlib/LibCURL`
  [76f85450] LibGit2 `@stdlib/LibGit2`
  [8f399da3] Libdl `@stdlib/Libdl`
  [56ddb016] Logging `@stdlib/Logging`
  [d6f4376e] Markdown `@stdlib/Markdown`
  [ca575930] NetworkOptions `@stdlib/NetworkOptions`
  [44cfe95a] Pkg `@stdlib/Pkg`
  [de0858da] Printf `@stdlib/Printf`
  [3fa0cd96] REPL `@stdlib/REPL`
  [9a3f8284] Random `@stdlib/Random`
  [ea8e919c] SHA `@stdlib/SHA`
  [9e88b42a] Serialization `@stdlib/Serialization`
  [6462fe0b] Sockets `@stdlib/Sockets`
  [fa267f1f] TOML `@stdlib/TOML`
  [a4e569a6] Tar `@stdlib/Tar`
  [8dfed614] Test `@stdlib/Test`
  [cf7118a7] UUIDs `@stdlib/UUIDs`
  [4ec0a83e] Unicode `@stdlib/Unicode`
  [deac9b47] LibCURL_jll `@stdlib/LibCURL_jll`
  [29816b5a] LibSSH2_jll `@stdlib/LibSSH2_jll`
  [c8ffd9c3] MbedTLS_jll `@stdlib/MbedTLS_jll`
  [14a3606d] MozillaCACerts_jll `@stdlib/MozillaCACerts_jll`
  [83775a58] Zlib_jll `@stdlib/Zlib_jll`
  [8e850ede] nghttp2_jll `@stdlib/nghttp2_jll`
  [3f19e933] p7zip_jll `@stdlib/p7zip_jll`
Precompiling project...
  ✗ FLOWExaFMM
  0 dependencies successfully precompiled in 1 seconds (8 already precompiled)
  1 dependency errored. To see a full report either run `import Pkg; Pkg.precompile()` or load the package
     Testing Running tests...
/local/disk1/pmandl/01_Software/julia-1.6.7/bin/julia: symbol lookup error: /local/disk1/pmandl/01_Software/FLOWExaFMM/src/fmm.so: undefined symbol: _ZN5jlcxx6Module12get_constantERKSs
ERROR: LoadError: Failed to precompile FLOWExaFMM [a07d1f4e-0e34-4d8b-bfef-e5b961477d34] to /home/pmandl/.julia/compiled/v1.6/FLOWExaFMM/jl_xaHY8X.
Stacktrace:
 [1] error(s::String)
   @ Base ./error.jl:33
 [2] compilecache(pkg::Base.PkgId, path::String, internal_stderr::Base.TTY, internal_stdout::Base.TTY, ignore_loaded_modules::Bool)
   @ Base ./loading.jl:1385
 [3] compilecache(pkg::Base.PkgId, path::String)
   @ Base ./loading.jl:1329
 [4] _require(pkg::Base.PkgId)
   @ Base ./loading.jl:1043
 [5] require(uuidkey::Base.PkgId)
   @ Base ./loading.jl:936
 [6] require(into::Module, mod::Symbol)
   @ Base ./loading.jl:923
 [7] include(fname::String)
   @ Base.MainInclude ./client.jl:444
 [8] top-level scope
   @ none:6
in expression starting at /local/disk1/pmandl/01_Software/FLOWExaFMM/test/runtests.jl:2
ERROR: Package FLOWExaFMM errored during testing

EdoAlvarezR, 2023-06-21:
Is the Hello World test mentioned here in the manual compilation passing? https://nbviewer.org/github/byuflowlab/FLOWVPM.jl/blob/master/docs/installation-linux.ipynb