byuflowlab/FLOWUnsteady #78: Actuator Surface Model with Propeller wing Interaction

inse0918, 2023-06-28:
Hi, I'm running the propeller wing interaction script.

But the issue is the code does not running.

I upload the script that run the propeller wing interaction code.

I used 'blownwing.jl' script with 'ASM.docs'.

I added the ASM script at blownwing.jl

But the code does not running...

Could you please check the code script ?

Thank you.

Best regards,
Inseo.

[PropellerWing-ASM.txt](https://github.com/byuflowlab/FLOWUnsteady/files/11891905/PropellerWing-ASM.txt)
[image]

EdoAlvarezR, 2023-08-17:
Hi @inse0918! This is a late reply and probably you already figured this out, but something to watchout for is that the PROWIM experiment used by Leo Veldhuis used a different thrust setting (CT=0.121 at J=0.85) that the what posterior studies reported with the isolated Beaver propeller. This leads to conclude that PROWIM used a collective pitch in the blade to achieve that thrust setting. See more details in [Sec. 8.4 of my dissertation](https://scholarsarchive.byu.edu/cgi/viewcontent.cgi?article=10598&context=etd):

[image]

Another thing: double check that the freestream matches the 49.5 m/s that they used in the experiment

EdoAlvarezR, 2024-01-31:
Resolved in #129

closed 2024-01-31 by EdoAlvarezR