byuflowlab/FLOWUnsteady #80: Rotor and Propeller Setup show for the same inputs different results

philippmandl, 2023-07-17:
Dear all, 

after running for a predefined propeller geometry, which was validated against CFD, the mid-fidelity "rotor"-setup (linked to the rotor-tutorial) with the same inputs, the thrust coefficienct is nearly devided by two. After changing some parameters linked to the "rotor" tutorial, the value doesn't change in a way that it is as close as to the "propeller" case. Is there anything I may miss in the setup?

Thanks for your response!

EdoAlvarezR, 2023-07-17:
Hi Philip! Would you mind sharing more details on your simulation? Particularly, how did you calculate the airfoil polars used in your blade geometry? 

The accuracy of the actuator line model hinges on having accurate airfoil polars used by the blade elements (very similar to a blade-element momentum method, except that the momentum part that represents the wake is provided by the VPM-resolved wake).

EdoAlvarezR, 2023-07-17:
Also, something to watch out for is that the definition of thrust coefficient CT in the rotorcraft community (CT= T / rho*pi*Omega*R^4) is different to the definition in the aircraft propeller community (CT = T / rho*n^2*D^4). FLOWUnsteady uses the aircraft propeller definition of CT: [LINK](https://flow.byu.edu/FLOWUnsteady/api/flowunsteady-monitor/#FLOWUnsteady.generate_monitor_rotors)

philippmandl, 2023-07-17:
Hi Ed, 
thank you very much for your quick reply! The only thing I changed ist the definition of the airfoil, where the polars are calculated within Xfoil - I set the command Xfoil=True

What I did so far is to implement my on my own created shape within the "Propeller" setup and the thrust and torque are compareable with the CFD results (I also used Xfoil for the polar creation) --> using the correct CT definition of the propeller community leads to compareable values . Implementing the same airfoils within the "rotor-hover" setup, but setting instead of 0.0001m/s inflow velocity the correct velocity of 30m/s and using the same definition of the CT as in the Propeller setup, leads to the fact that the overall thrust is too low compared to CFD results. Even changing the "low-fidelity" pre-settings of the rotor-setup to mid- or high-fidelity doesn't change anything. 

Thanks for your help and reply!

EdoAlvarezR, 2023-08-17:
Would you mind sharing more details about the operating conditions that you are trying to simulate? In particular, it would be useful to know the following:

* Tip Mach number (`M = pi*RPM/60 * D  / speedofsound`)
* Diameter/chord-based Reynolds number (`ReD = rho * sqrt( uinf^2 + (pi*RPM/60 * D)^2 ) * D / mu` )
* Advance ratio (`J = uinf / (RPM/60 * D)`)
* Solidity of the rotor

Something that you might want to try: while a rotor in hover needs tip and hub corrections to accurately account for the blade loading, it is not so for a propeller. Hence, [the rotor example](https://flow.byu.edu/FLOWUnsteady/examples/rotorhover-aero/) uses 
```julia
hubtiploss_correction = ((0.4, 5, 0.1, 0.05), (2, 1, 0.25, 0.05))
```
for your application I would recommend doing what [the propeller example](https://flow.byu.edu/FLOWUnsteady/examples/propeller-J040/) uses:
```julia
hubtiploss_correction = vlm.hubtiploss_nocorrection
```

EdoAlvarezR, 2023-08-17:
Also, have you tried visualizing the simulation and seeing if there's something blatantly wrong in the way the simulation was setup? (e.g., freestream pointing in the wrong direction, etc)

EdoAlvarezR, 2023-08-18:
Propeller simulations (rotor with axial freestream) are numerically very simple to simulate since the freestream helps the wake to deploy (your simulation can be quite coarse and still get the right answer). On the other hand, the hover case is as challenging as it gets: the wake deploys solely because of self-induced velocity, the rotor chops through its own wake at the beginning of the simulation, and tip vortices leapfrog and breakdown into turbulence close to the plane of rotation. The tip and hub loss correction is one of the elements that goes into getting an accurate hover simulation (otherwise you a lot of refinement to get the right loading towards the blade tip).

philippmandl, 2023-08-19:
Many thanks for your answers! If I understood you correctly, you would recommend to use the prop-model for studdiing any axial- and non-axial inflow cinditions (e.g. VTOL applications) and the rotor hover model just for the hover cas? Or could it be a mixture of both settings?

EdoAlvarezR, 2023-08-22:
My experience is that I've needed to shift to the hover model and fine discretization at advance ratios below `J=0.1` (where `J = uinf / (RPM/60 * D)`). For `J>0.2`, you are safe to use the propeller settings and very coarse discretization.

closed 2023-08-22 by EdoAlvarezR