byuflowlab/FLOWUnsteady #88: PyPlot version compatibility

cibinjoseph, 2023-08-25:
PyPlot v2.11.1 works.
PyPlot v2.11.2 does not.
Numpy v1.22.0 works, but newer versions might have an issue.

Seems to cause compilation errors for FLOWNoise.

Doing `] add PyPlot.jl` brought up this issue.

This is more of a note to self to investigate what causes it.

jackyw123, 2023-08-28:
So how do I make sure that the Pyplot version is v2.11.1 and how do I make sure that the Numpy version is v1.22.0?Thanks!

jackyw123, 2023-08-28:
I found that Pyplot and Numpy are both inside the matplotlib library, what can I do to make sure that I install the matplotlib library with Pyplot and Numpy versions v2.11.1 and v1.22.0?

vincinu, 2023-08-28:
] add PyPlot@2.11.1

jackyw123, 2023-08-29:
I often get the following exception display during the running of an example, what is the reason for this and how can I solve it?
[image]

EdoAlvarezR, 2023-09-05:
Could using this other package fix the issue?  [https://github.com/JuliaPy/PythonPlot.jl](https://github.com/JuliaPy/PythonPlot.jl)

AndreaRapi, 2023-10-24:
1. I installed and tested positive the FLOWExaFMM and FLOWVPM packages. 
2. Then I added the last packages as FLOWNoise and ran into errors:

```
`import` Pkg
url = "https://github.com/byuflowlab/"
packages = (("AirfoilPrep.jl", "v2.1.2"), ("FLOWVLM", "v2.1.2"), ("BPM.jl", "v2.0.1"), ("FLOWNoise", "v2.3.1"))
Pkg.add([ Pkg.PackageSpec(; url=url*name, rev=v) for (name, v) in packages ])
  ✗ AirfoilPrep
  ✗ FLOWNoise
  ✗ FLOWVLM
27 dependencies successfully precompiled in 115 seconds. 75 already precompiled. 1 skipped during auto due to previous errors. 
3 dependencies errored. To see a full report either run `import Pkg; Pkg.precompile()` or load the packages.**
```

3. I tried to add again the packages and tested the FLOWNoise package I ran into the following error:

julia> ] test FLOWNoise
Precompiling project...
  ✗ FLOWNoise
  0 dependencies successfully precompiled in 14 seconds. 76 already precompiled.
  1 dependency errored. To see a full report either run `import Pkg; Pkg.precompile()` or load the package
     Testing Running tests...
ERROR: LoadError: InitError: PyError (PyImport_ImportModule

The Python package matplotlib.pyplot could not be imported by pyimport. Usually this means that you did not install matplotlib.pyplot in the Python version being used by PyCall.

PyCall is currently configured to use the Python version at: /usr/bin/python3
and you should use whatever mechanism you usually use (apt-get, pip, conda, etcetera) to install the Python package containing the matplotlib.pyplot module.

One alternative is to re-configure PyCall to use a different Python version on your system: set ENV["PYTHON"] to the path/name of the python executable you want to use, run Pkg.build("PyCall"), and re-launch Julia.

Another alternative is to configure PyCall to use a Julia-specific Python distribution via the Conda.jl package (which installs a private Anaconda Python distribution), which has the advantage that packages can be installed and kept up-to-date via Julia.  As explained in the PyCall documentation, set ENV["PYTHON"]="", run Pkg.build("PyCall"), and re-launch Julia. Then, To install the matplotlib.pyplot module, you can use `pyimport_conda("matplotlib.pyplot", PKG)`, where PKG is the Anaconda package that contains the module matplotlib.pyplot, or alternatively you can use the Conda package directly (via `using Conda` followed by `Conda.add` etcetera).

I added here all the error lines: 
[FLOWNoise error.txt](https://github.com/byuflowlab/FLOWUnsteady/files/13111527/FLOWNoise.error.txt)

EdoAlvarezR, 2023-10-25:
Hi Andrea! Your error seems unrelated to the error discussed in this issue. The good news, though, is that it is really easy to fix =]

Did you follow the instructions under "PyCall" in the installation instructions? [flow.byu.edu/FLOWUnsteady/installation/general](https://flow.byu.edu/FLOWUnsteady/installation/general/#pycall)

The error that you are encountering means that your Python installation that Julia is calling through PyCall does not have the matplotlib library installed. To fix that, follow the instructions to make sure that PyCall is pointed to the right Python installation, then make sure you run the equivalent to the follow command that will install some dependencies into that Python installation:

```bash
pip3 install matplotlib mpmath scipy --user
```

AndreaRapi, 2023-10-26:
Great! I did all the installation instructions but the path was wrong. Now I fixed and I tried to the examples. Seems works.
[image]

EdoAlvarezR, 2024-04-16:
Issue solved [here](https://github.com/byuflowlab/FLOWNoise/issues/2#issuecomment-2060045775). Fix [is being merged](https://github.com/byuflowlab/FLOWUnsteady/pull/164) into master and release with new minor version.

closed 2024-04-16 by EdoAlvarezR