byuflowlab/FLOWUnsteady #90: Problem installing PyCall and running FLOWUnsteady program

DmitryKudryavtsevA, 2023-08-28:
There were problems installing PyCall, when checking the mpmath and scipy modules work correctly, matplotlib gives an error.

```
julia> import PyCall

julia> PyCall.pyversion
v"3.10.12"

julia> using PyCall

julia> math = pyimport("mpmath")
PyObject <module 'mpmath' from '/home/cfd/.local/lib/python3.10/site-packages/mpmath/__init__.py'>

julia> scp = pyimport("scipy")
PyObject <module 'scipy' from '/home/cfd/.local/lib/python3.10/site-packages/scipy/__init__.py'>

julia> plt = pyimport("matplotlib")
ERROR: PyError (PyImport_ImportModule

The Python package matplotlib could not be imported by pyimport. Usually this means
that you did not install matplotlib in the Python version being used by PyCall.

PyCall is currently configured to use the Julia-specific Python distribution
installed by the Conda.jl package.  To install the matplotlib module, you can
use `pyimport_conda("matplotlib", PKG)`, where PKG is the Anaconda
package that contains the module matplotlib, or alternatively you can use the
Conda package directly (via `using Conda` followed by `Conda.add` etcetera).

Alternatively, if you want to use a different Python distribution on your
system, such as a system-wide Python (as opposed to the Julia-specific Python),
you can re-configure PyCall with that Python.   As explained in the PyCall
documentation, set ENV["PYTHON"] to the path/name of the python executable
you want to use, run Pkg.build("PyCall"), and re-launch Julia.

) <class 'ModuleNotFoundError'>
ModuleNotFoundError("No module named 'pyparsing'")
  File "/home/cfd/.local/lib/python3.10/site-packages/matplotlib/__init__.py", line 129, in <module>
    from . import _api, _version, cbook, _docstring, rcsetup
  File "/home/cfd/.local/lib/python3.10/site-packages/matplotlib/rcsetup.py", line 28, in <module>
    from matplotlib._fontconfig_pattern import parse_fontconfig_pattern
  File "/home/cfd/.local/lib/python3.10/site-packages/matplotlib/_fontconfig_pattern.py", line 15, in <module>
    from pyparsing import (

Stacktrace:
 [1] pyimport(name::String)
   @ PyCall ~/.julia/packages/PyCall/ilqDX/src/PyCall.jl:558
 [2] top-level scope
   @ REPL[6]:1
```

Installing further according to the instructions and checking, I get the following error.

```
julia> import FLOWUnsteady as uns
[ Info: Precompiling FLOWUnsteady [b491798d-ac6e-455f-a27c-49c10bb0a666]
ERROR: LoadError: ArgumentError: hasproperty of NULL PyObject
Stacktrace:
 [1] pyhasproperty(o::PyCall.PyObject, s::String)
   @ PyCall ~/.julia/packages/PyCall/ilqDX/src/PyCall.jl:371
 [2] hasproperty
   @ ~/.julia/packages/PyCall/ilqDX/src/PyCall.jl:377 [inlined]
 [3] rc(args::String; kws::Base.Pairs{Symbol, String, Tuple{Symbol}, NamedTuple{(:family,), Tuple{String}}})
   @ PyPlot ~/.julia/packages/PyPlot/2MlrT/src/PyPlot.jl:191
 [4] top-level scope
   @ ~/.julia/packages/FLOWNoise/t5EXx/src/FLOWNoise.jl:38
 [5] include
   @ ./Base.jl:419 [inlined]
 [6] include_package_for_output(pkg::Base.PkgId, input::String, depot_path::Vector{String}, dl_load_path::Vector{String}, load_path::Vector{String}, concrete_deps::Vector{Pair{Base.PkgId, UInt64}}, source::String)
   @ Base ./loading.jl:1554
 [7] top-level scope
   @ stdin:1
in expression starting at /home/cfd/.julia/packages/FLOWNoise/t5EXx/src/FLOWNoise.jl:1
in expression starting at stdin:1
ERROR: LoadError: Failed to precompile FLOWNoise [d27480ee-285d-533b-ae3d-5018956ae0bc] to /home/cfd/.julia/compiled/v1.8/FLOWNoise/jl_ObeLaZ.
Stacktrace:
  [1] error(s::String)
    @ Base ./error.jl:35
  [2] compilecache(pkg::Base.PkgId, path::String, internal_stderr::IO, internal_stdout::IO, keep_loaded_modules::Bool)
    @ Base ./loading.jl:1707
  [3] compilecache
    @ ./loading.jl:1651 [inlined]
  [4] _require(pkg::Base.PkgId)
    @ Base ./loading.jl:1337
  [5] _require_prelocked(uuidkey::Base.PkgId)
    @ Base ./loading.jl:1200
  [6] macro expansion
    @ ./loading.jl:1180 [inlined]
  [7] macro expansion
    @ ./lock.jl:223 [inlined]
  [8] require(into::Module, mod::Symbol)
    @ Base ./loading.jl:1144
  [9] include
    @ ./Base.jl:419 [inlined]
 [10] include_package_for_output(pkg::Base.PkgId, input::String, depot_path::Vector{String}, dl_load_path::Vector{String}, load_path::Vector{String}, concrete_deps::Vector{Pair{Base.PkgId, UInt64}}, source::Nothing)
    @ Base ./loading.jl:1554
 [11] top-level scope
    @ stdin:1
in expression starting at /home/cfd/.julia/packages/FLOWUnsteady/Jl5A6/src/FLOWUnsteady.jl:1
in expression starting at stdin:1
ERROR: Failed to precompile FLOWUnsteady [b491798d-ac6e-455f-a27c-49c10bb0a666] to /home/cfd/.julia/compiled/v1.8/FLOWUnsteady/jl_cZ5DvK.
Stacktrace:
 [1] error(s::String)
   @ Base ./error.jl:35
 [2] compilecache(pkg::Base.PkgId, path::String, internal_stderr::IO, internal_stdout::IO, keep_loaded_modules::Bool)
   @ Base ./loading.jl:1707
 [3] compilecache
   @ ./loading.jl:1651 [inlined]
 [4] _require(pkg::Base.PkgId)
   @ Base ./loading.jl:1337
 [5] _require_prelocked(uuidkey::Base.PkgId)
   @ Base ./loading.jl:1200
 [6] macro expansion
   @ ./loading.jl:1180 [inlined]
 [7] macro expansion
   @ ./lock.jl:223 [inlined]
 [8] require(into::Module, mod::Symbol)
   @ Base ./loading.jl:1144
```

Installed according to these instructions.  https://flow.byu.edu/FLOWUnsteady/installation/general/#installation

vincinu, 2023-08-28:
I was using FLOWUnsteady with no issues until last week. Today I had this same error and I do not understand why. I installed XML, FileIO and LibXML2 packages in Julia. I thought it might be related to that. but after removing them and reinstalling all packages I still get the error

EdoAlvarezR, 2023-08-28:
Same than Issue #88 (see that thread for how to solve it)

closed 2023-08-28 by EdoAlvarezR