byuflowlab/FLOWUnsteady #101: FLOWNoise compiling error

dil937, 2023-09-18:
When installing "Other Packages" in the installation manual, FLOWNoise doesn't get compiled. It gives the following error. 

✗ FLOWNoise
  31 dependencies successfully precompiled in 51 seconds (72 already precompiled)
  1 dependency errored. To see a full report either run `import Pkg; Pkg.precompile()` or load the package

Any idea how this can be solved? Thank you so much for any help.

EdoAlvarezR, 2023-10-04:
This should have been fixed with the latest commit to Main in FLOWNoise. Could try adding FLOWNoise again and seeing if that fixed it?

tobiokanobi, 2023-11-08:
I just ran into the same problem. Changing pyplot to 2.11.1 resolved this issue

] add PyPlot@2.11.1

dil937, 2023-11-28:
Thanks! It worked!

closed 2023-11-28 by dil937

dingzhw, 2024-04-17:
> I just ran into the same problem. Changing pyplot to 2.11.1 resolved this issue
> 
> ] add PyPlot@2.11.1

Thank you, @tobiokanobi ,It works!