byuflowlab/FLOWUnsteady #103: Restart the simulation from an existing time step

philippmandl, 2023-09-28:
Dear All, 

I tried to restart one simulation from an existing time-step, from where the simulation should proceed on. What is currently happening, that I am facing an error which says in the end, that the keyword "result_file" is unsupported --> do you have any ideas or experience with restarting simulations? 

Many thanks!

EdoAlvarezR, 2023-09-28:
Hi Philip!

As indicated [in the docs](https://flow.byu.edu/FLOWUnsteady/api/flowunsteady-run/), the name of the keyword is `restart_vpmfile`. You could also check that directly in the code here: [LINK](https://github.com/byuflowlab/FLOWUnsteady/blob/master/src/FLOWUnsteady_simulation.jl#L165-L167)

closed 2023-09-28 by EdoAlvarezR

philippmandl, 2023-09-28:
great, thank you very much!!