byuflowlab/FLOWUnsteady #104: Propeller Geometry - Blade airfoil sections are input in flat (planar) sections rather than radial sections

surajrkashyap, 2023-10-04:
On a propeller, when I input the airfoil sections at different radial positions FLOWUnsteady seems to take them as planar sections (rather than radial sections concentric to the propeller axis). I am attaching a comparison below of the blade CAD and the Paraview ouput from FLOWUnsteady (not to scale). Am I missing anything obvious? Thanks for taking a look!

[image]

PS. This is a marine propeller. I noticed that the validation references (E. J. Alvarez & A. Ning (2020), "High-Fidelity Modeling of Multirotor Aerodynamic Interactions for Aircraft Design," AIAA Journal.) mention a similar marine propeller (INSEAN 779a). Was there any workaround used for the geometry?

EdoAlvarezR, 2023-10-20:
Hi! 

The blade definition assumes that the user will input airfoil section from the Cartesian reference system rather than a streamtube (or meridonial) reference plane. FLOWUnsteady was developed for aircraft rotor/propellers and wind turbines, which typically have a low solidity distribution away from the root, however, the numerical method should work relatively well for marine applications.

I'm sure others would appreciate you developing the capabilities of defining the blade from those streamtube reference planes!

closed 2024-04-17 by surajrkashyap

sakurakikiki, 2024-08-22:
> 你好！
> 
> 叶片定义假设用户将从笛卡尔参考系统输入翼型截面，而不是从流管（或经纬）参考平面输入翼型截面。FLOWUnsteady是为飞机转子/螺旋桨和风力涡轮机开发的，它们通常具有远离根部的低固体度分布，但是，数值方法应该适用于海洋应用。
> 
> 我相信其他人会感谢您从那些流管参考平面定义叶片的能力！

Hello, there is calculation of underwater propeller E779a in your literature, I would like to ask you why I calculated the state of constant divergence of underwater propeller