byuflowlab/FLOWUnsteady #108: question about the fdom.jl

inse0918, 2023-10-25:
Hello.

I ran the fdom.jl code for visualizing the fluid domain.

I set the velocity condition 49.7m/s (in wing.jl), but the results of the fdom is here.

[image]

The velocity field(x components) is lower than a 49.7 m/s which is meaning the free stream condition.

Is it right results ?.

Thank you.
Best regards,
Inseo.

*
The testing analysis code is example/wing/wing.jl

And the fdom code is here.

```
import FLOWUnsteady as uns
import FLOWUnsteady: vpm, gt, dot, norm

# --------------- INPUTS AND OUTPUTS -------------------------------------------
# INPUT OPTIONS
simulation_name = "wing-example"               # Simulation to read
read_path       = "/home/inse0918/Programs/FlowUnsteady/"*simulation_name # Where to read simulation from

pfield_prefix   = "wing-example_pfield"      # Prefix of particle field files to read
staticpfield_prefix = "wing-example_staticpfield" # Prefix of static particle field files to read

nums            = collect(100:1:110)           # Time steps to process

# OUTPUT OPTIONS
save_path       = joinpath(read_path, "..", simulation_name*"_post_fdom")  # Where to save fluid domain
output_prefix   = "wing-example_fdom"             # Prefix of output files
prompt          = true                      # Whether to prompt the user
verbose         = true                      # Enable verbose
v_lvl           = 0                         # Verbose indentation level

# -------------- PARAMETERS ----------------------------------------------------
# Simulation information
# R               = 11/39.37                      # (m) rotor radius
R               = 1.5                      # (m) rotor radius
AOA             = 0.0                       # (deg) angle of attack or incidence angle

# Grid
L               = R                         # (m) reference length
dx, dy, dz      = L/30, L/30, L/30          # (m) cell size in each direction
Pmin            = L*[-2, -2, -2]   # (m) minimum bounds
Pmax            = L*[ 5,  2,  2]   # (m) maximum bounds
NDIVS           = ceil.(Int, (Pmax .- Pmin)./[dx, dy, dz])  # Number of cells in each direction
nnodes          = prod(NDIVS .+ 1)          # Total number of nodes

Oaxis           = gt.rotation_matrix2(0, 0, AOA)    # Orientation of grid

# VPM settings
maxparticles    = Int(1.0e6 + nnodes)         # Maximum number of particles
fmm             = vpm.FMM(; p=4, ncrit=50, theta=0.4, phi=0.3) # FMM parameters (decrease phi to reduce FMM noise)
scale_sigma     = 1.00                      # Shrink smoothing radii by this factor
f_sigma         = 0.5                       # Smoothing of node particles as sigma = f_sigma*meansigma

maxsigma        = L/10                      # Particles larger than this get shrunk to this size (this helps speed up computation)
maxmagGamma     = Inf                       # Any vortex strengths larger than this get clipped to this value

include_staticparticles = true              # Whether to include the static particles embedded in the solid surfaces

other_file_prefs = include_staticparticles ? [staticpfield_prefix] : []
other_read_paths = [read_path for i in 1:length(other_file_prefs)]

if verbose
    println("\t"^(v_lvl)*"Fluid domain grid")
    println("\t"^(v_lvl)*"NDIVS =\t$(NDIVS)")
    println("\t"^(v_lvl)*"Number of nodes =\t$(nnodes)")
end

# --------------- PROCESSING SETUP ---------------------------------------------
if verbose
    println("\t"^(v_lvl)*"Getting ready to process $(read_path)")
    println("\t"^(v_lvl)*"Results will be saved under $(save_path)")
end

# Create save path
if save_path != read_path
    gt.create_path(save_path, prompt)
end

# Generate function to process the field clipping particle sizes
preprocessing_pfield = uns.generate_preprocessing_fluiddomain_pfield(maxsigma, maxmagGamma;
                                                                        verbose=verbose, v_lvl=v_lvl+1)

# --------------- PROCESS SIMULATION -------------------------------------------

nthreads        = 1                         # Total number of threads
nthread         = 1                         # Number of this thread
dnum = floor(Int, length(nums)/nthreads)    # Number of time steps per thread
threaded_nums = [view(nums, dnum*i+1:(i<nthreads-1 ? dnum*(i+1) : length(nums))) for i in 0:nthreads-1]

for these_nums in threaded_nums[nthread:nthread]

     uns.computefluiddomain(    Pmin, Pmax, NDIVS,
                                maxparticles,
                                these_nums, read_path, pfield_prefix;
                                Oaxis=Oaxis,
                                fmm=fmm,
                                f_sigma=f_sigma,
                                save_path=save_path,
                                file_pref=output_prefix, grid_names=["_fdom"],
                                other_file_prefs=other_file_prefs,
                                other_read_paths=other_read_paths,
                                userfunction_pfield=preprocessing_pfield,
                                verbose=verbose, v_lvl=v_lvl)
end
```

EdoAlvarezR, 2023-10-25:
Hi Inseo, 

The `computefluiddomain(...)` function receives [the optional argument `add_Uinf`](https://github.com/byuflowlab/FLOWVPM.jl/blob/master/examples/utilities/utilities_fluiddomain.jl#L30) indicating whether to add the freestream to the computed fluid domain. The default value is `add_Uinf=false`, which means that the fluid domain that is generated is only what is induced by the particle field (free particles in the wake, and embedded particles on the surface if you are also giving it the files with static particles).

To have the freestream added to the fluid domain, you can either do `computefluiddomain(...; add_Uinf=true, ...)` or you can also add the freestream velocity manually inside ParaView with a Calculator filter.

closed 2023-10-25 by EdoAlvarezR

inse0918, 2023-10-26:
> Collaborator

Hi, Alvarez.

Thank you for your kind response even though you are busy nowadays :).

I have another question about post-processing the forces generated by propeller.

The thrust coefficient(CT) which means the axial thrust coefficient(CTx) when we see the post-processed data like *.csv and *.convergence plot by png.

How can I export the other components about CT(ex. CTy: side force, CTz: normal force) ?

Thank you.

Best regards,
Inseo.

EdoAlvarezR, 2023-10-26:
The full force vector along the blade is exported with the vtk blade files. You can process then postprocess the blade in ParaView to integrate and decompose the force however you need, or you can also do that programmatically reading these vtk files in Julia (that is what the hover tutorial does [LINK1](https://flow.byu.edu/FLOWUnsteady/examples/rotorhover-aero/) [LINK2](https://github.com/byuflowlab/FLOWUnsteady/blob/master/examples/rotorhover/rotorhover_postprocessing.jl#L127) )