byuflowlab/FLOWUnsteady #110: compile ExaFMM running the script build.sh

dadouk, 2023-11-01:
When try to compile  ExaFMM running the script build.sh I got an error message:

"configure: error: could not find mpi library for --enable-mpi
Compiling 3d
make: *** No targets specified and no makefile found.  Stop.
cp: cannot stat 'build/3d/fmm': No such file or directory
Done!"

and the file fmm.so is not generated

Any solutions?

cibinjoseph, 2024-02-05:
Sorry, I've had this error and couldn't figure out a fix for it. I guess it has to do with the path to mpi.h and similar files not being set correctly somewhere. Typically, this can be corrected by initializing the environment variables or the path. But, I haven't got it working yet.

davidosterberg, 2025-03-21:
This likely happens because you need to install 'openmpi'

closed 2026-03-06 by BTV25