byuflowlab/FLOWUnsteady #111: Simulating the ground effect

ss-hegde, 2023-11-03:
Hello all,

I am trying to simulate the propeller wake-ground interactions. I could find a function which generates the ground surface. 
```
"""
    Generates ground surface.
"""
function generate_ground_vahana(bscale; verify_spline=false, spl_s=0.001,
                                spl_k="automatic", visualize=false)
```
How should I define this ground in the simulation, so that the wake interacts with the ground surface?

Thanks in advance!

weijunsjtu, 2023-12-21:
Hello, have you solved this problem yet. How did you define this ground function. Very much looking forward to your answer

EdoAlvarezR, 2024-01-31:
The capabilities for modeling the ground with a panel method are available in the [`flowpanel-prototype` branch](https://github.com/byuflowlab/FLOWUnsteady/tree/flowpanel-prototype) but no documentation or examples are yet provided for users since these features are still under heavy development.