byuflowlab/FLOWUnsteady #113: CxxWrap required version fails to compile on Julia 1.9

cibinjoseph, 2023-11-12:
CxxWrap.jl version 0.11.2 required for FLOWUnsteady does not compile on Julia 1.9 with the error:
```
ERROR: LoadError: libcxxwrap_julia_jll not available on this platform
Stacktrace:
 [1] error(s::String)
   @ Base ./error.jl:35
 [2] top-level scope
   @ ~/.julia/packages/CxxWrap/OcN1Z/src/CxxWrap.jl:18
 [3] include
   @ ./Base.jl:457 [inlined]
 [4] include_package_for_output(pkg::Base.PkgId, input::String, depot_path::Vector{String}, dl_load_path::Vector{String}, load_path::Vector{String}, concrete_deps::Vector{Pair{Base.PkgId, UInt128}}, source::Nothing)
   @ Base ./loading.jl:2049
 [5] top-level scope
```
This is because `libcxxwrap_julia_jll` is not available in Julia 1.9. The only workaround is to use Julia 1.8.
Refererence discussion [thread](https://github.com/JuliaInterop/CxxWrap.jl/issues/386#issuecomment-1779968340)

EdoAlvarezR, 2024-04-15:
Has anyone looked at what it would take to get things working with Julia v1.9 or newer?

EdoAlvarezR, 2024-07-20:
Fixed. Now it's working with Julia v1.10+.

closed 2024-07-20 by EdoAlvarezR