byuflowlab/FLOWUnsteady #114: how to define a rotor with polars as a function of Mach number

eversilia, 2023-11-12:
Hi, 
can I define a blade with multiple airfoil polars that describe CL,CD and CM as a function of AoA **and Mach** ?
In the rotor example, the blade is defined without taking into account compressibility effects as far as I can tell !
Thank you in advance !

cibinjoseph, 2023-12-01:
FLOWUnsteady has the provision to receive individual airfoil data tables for each section. You could provide separate Mach corrected airfoil data for each section.

eversilia, 2023-12-01:
Thank you for your reply !
That is for sure duable for a propeller or in a rotor hovering condition.
On the other hand, in forward flight, the single section (depending on its position) can experience big variation in mach number. In this case we need the code to automatically query coefficients depending on the instantaneous Mach.

cibinjoseph, 2023-12-01:
Ah, that's right. Should be possible to include that provision down the line as the code matures.

EdoAlvarezR, 2024-01-31:
Related to #136.

As Cibin said, the capabilities of re-calculating (or providing) polars throughout the maneuver might be developed in the future.