byuflowlab/FLOWUnsteady #115: Acoustic Analysis Error

philippmandl, 2023-11-17:
Dear All, 

I am currently running acoustic postprocessing which is working quite well until the following error occours:

julia> # Read datasets and stores them in dictionaries
       noise.read_data(dataset_infos; datasets_pww=datasets_pww, datasets_bpm=datasets_bpm)
************************************************************************
*    Reading dataset /xy/example
************************************************************************
    Reading output file pressure.nam ...
        Found 72x1 grid in WOPWOP output with 1440 time steps.
    Reading output file segmentProcess/spl_spectrum.nam ...
        Found 72x1 grid in WOPWOP output with 721 time steps.
ERROR: MethodError: Cannot `convert` an object of type Symbol to an object of type Float64
Closest candidates are:
  convert(::Type{T}, ::ColorTypes.Gray24) where T<:Real at ~/.julia/packages/ColorTypes/1dGw6/src/conversions.jl:114
  convert(::Type{T}, ::ColorTypes.Gray) where T<:Real at ~/.julia/packages/ColorTypes/1dGw6/src/conversions.jl:113
  convert(::Type{T}, ::Base.TwicePrecision) where T<:Number at twiceprecision.jl:273
  ...
Stacktrace:
 [1] setindex!(::Array{Float64, 4}, ::Symbol, ::Int64, ::Int64, ::Int64, ::Vararg{Int64})
   @ Base ./array.jl:968
 [2] read_wopwopoutput(outputname::String; read_path::String, verbose::Bool, verbose_level::Int64, v_lvl::Int64, tec::Bool)
   @ FLOWNoise ~/.julia/packages/FLOWNoise/t5EXx/src/FLOWNoise_wopwop.jl:120
 [3] #read_pwwfield#63
   @ ~/.julia/packages/FLOWNoise/t5EXx/src/FLOWNoise_wopwop.jl:943 [inlined]
 [4] read_pwwfield(fieldnames::Vector{String}, read_path::String; re20crit::Vector{String}, tec::Bool, optargs::Base.Pairs{Symbol, Integer, Tuple{Symbol, Symbol, Symbol}, NamedTuple{(:verbose, :verbose_level, :v_lvl), Tuple{Bool, Int64, Int64}}})
   @ FLOWNoise ~/.julia/packages/FLOWNoise/t5EXx/src/FLOWNoise_wopwop.jl:964
 [5] fetch_pwwfield(::String; fieldnames::Vector{String}, datasets::Dict{Any, Any}, verbose::Bool, v_lvl::Int64, verbose_level::Int64, optargs::Base.Pairs{Symbol, Union{}, Tuple{}, NamedTuple{(), Tuple{}}})
   @ FLOWNoise ~/.julia/packages/FLOWNoise/t5EXx/src/FLOWNoise_wopwop.jl:983
 [6] fetch_pwwdataset(::String; datasets::Dict{Any, Any}, verbose::Bool, v_lvl::Int64, force_read::Bool, optargs::Base.Pairs{Symbol, Vector{String}, Tuple{Symbol}, NamedTuple{(:fieldnames,), Tuple{Vector{String}}}})
   @ FLOWNoise ~/.julia/packages/FLOWNoise/t5EXx/src/FLOWNoise_wopwop.jl:998
 [7] read_data(dataset_infos::Vector{NTuple{5, String}}; datasets_pww::Dict{Any, Any}, datasets_bpm::Dict{Any, Any}, pww_fieldnames::Vector{String}, bpm_fieldnames::Vector{String}, optargs::Base.Pairs{Symbol, Union{}, Tuple{}, NamedTuple{(), Tuple{}}})
   @ FLOWNoise ~/.julia/packages/FLOWNoise/t5EXx/src/FLOWNoise_postprocessing.jl:64
 [8] top-level scope
   @ REPL[405]:2

do you have an idea how I can fix this issue? 

Many thanks in advance!

philippmandl, 2023-11-23:
problem has been solved

philippmandl, 2023-11-23:
solved

closed 2023-11-23 by philippmandl

EvenW254, 2025-01-12:
@ilm71 Hello!
Could you please send me a copy of the PSU-WOPWOP binary executable code？
Thank U！