byuflowlab/FLOWUnsteady #120: Tutorials - Rotorhover

AndreaRapi, 2023-12-11:
Hi there,

I proceeded to simulate this tutorial as follow:
1. I did the "Variable Fidelity" part for low and mid-high cases.
2. I have implement and did the "Fluid Domain" part related to the mid-high case.
3. I have implemented the aeroacustics part but I get the following error:
`IOError: open("/home/edoalvar/Dropbox/WhisperAero/OtherCodes/PSU-WOPWOP_v3.4.4/wopwop3_linux_serial", 0, 0):`

Thank you in advance,
Andrea.

[rotorhover_aeroacustics_error.txt](https://github.com/byuflowlab/FLOWUnsteady/files/13633714/rotorhover_aeroacustics_error.txt)

JonathonBowen, 2023-12-12:
Hi Andrea, 

To run the aeroacoustics package you require a copy of PSU-WOPWOP (an FW-H code that was written at Penn State) which is not included by FLOWUnsteady. You can request a copy of the binary by contacting them directly. 

Once you have copy you will have to change your input file so that it is looking in the correct directory for the binary. 

Hope this helps.

Cheers

AndreaRapi, 2023-12-14:
Hi Jonathon,

Many thanks for the info. Do you have the binary code or you know how is the reference in Penn State?

Cheers,
Andrea