byuflowlab/FLOWUnsteady #124: PSU-WOPWOP code request

AndreaRapi, 2023-12-20:
Hi there,

I tried to ask the copy of the PSU-WOPWOP code without any success. I contacted both the Penn State professor, his colleagues, and colleagues who used the code but the responses did not provide me with the PSU-WOPWOP code. May I ask for your kind help?

Thank you in advance,
Andrea.

zyfporsche, 2024-01-03:
> Hi there,
> 
> I tried to ask for a copy of the PSU-WOPWOP code without any success. I contacted both the Penn State professor, his colleagues, and colleagues who used the code but the responses did not provide me with the PSU-WOPWOP code. May I ask for your kind help?
> 
> Thank you in advance, Andrea.

Hi, bro, do you have a binary copy of PSU-WOPWOP? I can't even find a binary one. 
I don't know why but I can't log in this website. (https://sites.psu.edu/wopwop/)

AndreaRapi, 2024-01-04:
Hi,

I don't know about this website, I am still waiting the response from the Penn State University.

closed 2024-01-18 by AndreaRapi

NM1293, 2024-04-11:
> Hi,
> 
> I don't know about this website, I am still waiting the response from the Penn State University.

Hi Andrea,

I also wrote to Professor Brentner in the past few days to obtain the code, but I haven't received any response yet. Did you manage to obtain the PSU-WOPWOP binary instead?

Thank you.

AndreaRapi, 2024-04-11:
@NM1293 Yes, but I had to insist a lot, with several emails.

NM1293, 2024-04-11:
> @NM1293 Yes, but I had to insist a lot, with several emails.

It's a problem to send the code to this email: woviw93868@eryod.com?

Thank you in any case.

NM1293, 2024-04-11:
I can imagine this, but I hope that Prof. Brentner grants permission to distribute his code when he sends it.

Thank you, Andrea.

Have a nice day and good work with FLOWUnsteady.

VE111, 2024-12-17:
@NM1293 Hi！
Excuse me, have you already obtained the code? Could you please send me a copy?
Have a nice day!