byuflowlab/FLOWUnsteady #125: Loading (N/m) calculation of Propeller

philippmandl, 2023-12-21:
Dear All, 

currently I am comparing my BEMT results to FlowUnsteady by comparing the Propeller loading (N/m) 

[image]

How was the loading of the validation case calculated? I have implemented in the "run_simulation" the "debug=true" line to gain any information out of the .vtk files. 

Thank you very much for your answer! 

Best

EdoAlvarezR, 2023-12-21:
Hi Philipp!

The "Rotor Hover" tutorial has an example on how to read the blade loading from the VTK files and take a time average:

* [Tutorial](https://flow.byu.edu/FLOWUnsteady/examples/rotorhover-aero/)
* [Code](https://github.com/byuflowlab/FLOWUnsteady/blob/master/examples/rotorhover/rotorhover_postprocessing.jl)

philippmandl, 2023-12-21:
Hi Ed, 

many thanks for your quick reply. So as far I can understand, the blade loading is an average loading of the fluctuations on the blade caused by the rotations? 

My first intention was, that the blade loading is the "Thrust[i]/chord[i] "distribution

philippmandl, 2023-12-21:
And compared to that: What is meant by the  "Distributed_Load_Magnitude"? How can that be expressed?

[image]

EdoAlvarezR, 2023-12-21:
Blade loading (or distributed load) is force per unit span, hence it has unit of N/m. The simulation computes the instantaneous loading in each time step, which depending on your case it might fluctuate in time (due to blade vortex interaction, edgewise flow, etc), hence the need of time averaging in.

EdoAlvarezR, 2023-12-21:
Be aware that the `DistributedLoad` vector along the blade contains both the normal and tangential force. The rotor thrust comes from integrating the normal force over each blade, while rotor torque is computed from integrating position x tangential force over each blade. In the literature (and in the figure that you are asking about), blade loading is the distributed normal force.

philippmandl, 2023-12-21:
Okay thanks for the input - I was just wondering that if I devide the DistributedLoad by 1m(1000mm), I have for a hover propeller the same distribution as for the C_t distribution from my BEMT.

philippmandl, 2023-12-21:
and therefore I am wondering if this is reasonable - what do you think?

EdoAlvarezR, 2023-12-21:
The distributed load is the force already divided by the span length, so it will end up with the wrong units. It also depends on how you are defining C_t in your BEMT.

philippmandl, 2023-12-21:
Within the BEMT, C_t is defined based on the sectional Thrust as a function of (phi) -->  (Lift*cos(phi)-Drag*sin(phi)), where phi is the angle between the free-stream velocity and the propeller plane

philippmandl, 2023-12-21:
But when des DistributedLoad is the Normal-Force (I guess normal to the propeller plane), the Loading should be equivalent to the thrust distribution in hover

EdoAlvarezR, 2023-12-21:
> Be aware that the DistributedLoad vector along the blade contains both the normal and tangential force.
You'll have to decompose DistributedLoad into the direction that makes the normal force to get the thrust distribution.

From the normal force distribution you can then compute C_t in a manner that is consistent with your BEMT outputs (probably `C_t = N / (0.5 * rho * Uinf^2 * chord)` where `N` is the normal force at that station, perhaps multiplied by the number of blades depending on how your BEMT defined its outputs).

Alternatively, you can follow [this discussion](https://github.com/byuflowlab/FLOWUnsteady/discussions/96) to output the tangential (`ct`) and normal (`cn`) force coefficients that were calculated internally, which probably resemble the definition of your BEMT (see [this](https://github.com/byuflowlab/FLOWVLM/blob/master/src/FLOWVLM_rotor.jl#L2083C1-L2087C49))

philippmandl, 2023-12-22:
Thanks for your quick feedback - I have already plotted the internaly C_t values at a specific time step and compared this value with my C_t BEMT Results at the predefined blade sections - the results are not compareable, but the overall thrust and torque are matching exactly. 

That's why I am wondering where the error could be

EdoAlvarezR, 2023-12-22:
'ct' in FLOWUnsteady is the tangential force coefficient, while 'cn' is the normal force coefficient. My guess is that C_t in your BEMT is a "thrust" coefficient, so you actually want to compare C_t against 'cn'. Also, it's possible that your BEMT already multiplied the coefficient by the number of blades.

philippmandl, 2023-12-22:
Thanks Ed for the answer - I have already plotted the results previously and the "c_n" value gained within FlowUnsteady is slightely higher compared to my BEMT results -nevertheless, the plot looks compareable ... the overall Thrust and Torque are matching quite well too, now I have to figure out why FlowUnsteady predicts higher C_n values than the BEMT

EdoAlvarezR, 2023-12-22:
The main difference is that FLOWUnsteady is a wake-resolving simulation capturing contraction of the streamtube and all other wake dynamics (these become particularly important in near hover conditions). BEMT on the other hand simply assumes a straight semi-infinite cylindrical streamtube (which is baked into the momentum balance part of the theory that leads to the far field induced velocities being double the near field induced velocities).

With that in mind, you should expect FLOWUnsteady to be more accurate than BEMT.

EdoAlvarezR, 2023-12-22:
See this VPM vs experimental vs URANS vs BEMT comparison in the tutorials: https://flow.byu.edu/FLOWUnsteady/examples/rotorhover-aero/

philippmandl, 2023-12-22:
Yes, that‘s why I am
thinking about the possible error - I am not using the BEMT implemented within the Flowunsteady code but using my own BEMT model, therefore differenced may occur…

philippmandl, 2023-12-27:
> From the normal force distribution you can then compute C_t in a manner that is consistent with your BEMT outputs (probably `C_t = N / (0.5 * rho * Uinf^2 * chord)` where `N` is the normal force at [...]

I guess for the chord you used the mean-chordlength?