byuflowlab/FLOWUnsteady #126: Fluid domain visualization

MarcoPic99, 2023-12-22:
Hi everyone!

I started using FLOWUnsteady a few days ago and encountered some difficulties in displaying the fluid domain in Paraview. In particular, when loading the 'fdom' file generated as output by the 'computefluiddomain' function in Paraview, I notice that the 'Bounds' show x, y, and z ranges equal to 0 (as shown in the attached image). I also tried simulating the 'Rotor in Hover' example, but in this case,
the same issue occurs. However, in the .pvsm file provided in the tutorial, the x, y, and z ranges are
different from 0 and match the grid size defined in the 'computefluiddomain' function.
So, I wanted to ask if this is an issue with the input to the 'computefluiddomain' function or if it requires some action within Paraview.

Below the computefluidomain function I used and attached the .xmf file of the fluid domain
[blownwing_fdom.100.txt](https://github.com/byuflowlab/FLOWUnsteady/files/13753882/blownwing_fdom.100.txt)
 and a figure of the bounds i see in paraview.
[image]

Thank you in advance!

Marco

```
`#=##############################################################################
   
=###############################################################################

import FLOWUnsteady as uns
import FLOWUnsteady: vpm, gt, dot, norm

# --------------- INPUTS AND OUTPUTS -------------------------------------------
# INPUT OPTIONS
simulation_name = "blownwing"               # Simulation to read
read_path       = "save_propIsolata/" #*simulation_name # Where to read simulation from

pfield_prefix   = "blownwing_pfield"      # Prefix of particle field files to read
staticpfield_prefix = "blownwing_staticpfield" # Prefix of static particle field files to read

nums            = [100]              # Time steps to process

# OUTPUT OPTIONS
save_path       = joinpath(read_path, "..", simulation_name*"-fdom")  # Where to save fluid domain
output_prefix   = "blownwing"             # Prefix of output files
prompt          = true                      # Whether to prompt the user
verbose         = true                      # Enable verbose
v_lvl           = 0                         # Verbose indentation level

# -------------- PARAMETERS ----------------------------------------------------
# Simulation information
R               = 0.187                      # (m) rotor radius
AOA             = 0.0                       # (deg) angle of attack or incidence angle

# Grid
L               = R                         # (m) reference length
dx, dy, dz      = L/50, L/50, L/50          # (m) cell size in each direction
Pmin            = L*[-1.25, -0.50, -1.25]   # (m) minimum bounds
Pmax            = L*[ 1.25,  2.0,  1.25]   # (m) maximum bounds
NDIVS           = ceil.(Int, (Pmax .- Pmin)./[dx, dy, dz])  # Number of cells in each direction
nnodes          = prod(NDIVS .+ 1)          # Total number of nodes

Oaxis           = gt.rotation_matrix2(0, 0, AOA)    # Orientation of grid

# VPM settings
maxparticles    = Int(1.0e6 + nnodes)         # Maximum number of particles
fmm             = vpm.FMM(; p=4, ncrit=50, theta=0.4, phi=0.3) # FMM parameters (decrease phi to reduce FMM noise)
scale_sigma     = 1.00                      # Shrink smoothing radii by this factor
f_sigma         = 0.5                       # Smoothing of node particles as sigma = f_sigma*meansigma

maxsigma        = L/10                      # Particles larger than this get shrunk to this size (this helps speed up computation)
maxmagGamma     = Inf                       # Any vortex strengths larger than this get clipped to this value

include_staticparticles = true              # Whether to include the static particles embedded in the solid surfaces

other_file_prefs = include_staticparticles ? [staticpfield_prefix] : []
other_read_paths = [read_path for i in 1:length(other_file_prefs)]

if verbose
    println("\t"^(v_lvl)*"Fluid domain grid")
    println("\t"^(v_lvl)*"NDIVS =\t$(NDIVS)")
    println("\t"^(v_lvl)*"Number of nodes =\t$(nnodes)")
end

# --------------- PROCESSING SETUP ---------------------------------------------
if verbose
    println("\t"^(v_lvl)*"Getting ready to process $(read_path)")
    println("\t"^(v_lvl)*"Results will be saved under $(save_path)")
end

# Create save path
if save_path != read_path
    gt.create_path(save_path, prompt)
end

# Generate function to process the field clipping particle sizes
preprocessing_pfield = uns.generate_preprocessing_fluiddomain_pfield(maxsigma, maxmagGamma;
                                                                        verbose=verbose, v_lvl=v_lvl+1)

# --------------- PROCESS SIMULATION -------------------------------------------

nthreads        = 1                         # Total number of threads
nthread         = 1                         # Number of this thread
dnum = floor(Int, length(nums)/nthreads)    # Number of time steps per thread
threaded_nums = [view(nums, dnum*i+1:(i<nthreads-1 ? dnum*(i+1) : length(nums))) for i in 0:nthreads-1]

for these_nums in threaded_nums[nthread:nthread]

     uns.computefluiddomain(    Pmin, Pmax, NDIVS,
                                maxparticles,
                                these_nums, read_path, pfield_prefix;
                                Oaxis=Oaxis,
                                fmm=fmm,
                                f_sigma=f_sigma,
                                save_path=save_path,
                                file_pref=output_prefix, grid_names=["_fdom"],
                                other_file_prefs=other_file_prefs,
                                other_read_paths=other_read_paths,
                                userfunction_pfield=preprocessing_pfield,
                                verbose=verbose, v_lvl=v_lvl)

end'
```

EdoAlvarezR, 2023-12-30:
Hi Marco! 
Could you please upload both `.xmf` and `.h5` files?

MarcoPic99, 2024-01-01:
Hello, thank you very much for the answer, here is the link to the files.

https://drive.google.com/drive/folders/1EeqZ_cTeOZ1Z94AjD0AfpYA-GxeXM942?usp=drive_link

And happy new year!

closed 2024-01-29 by MarcoPic99