byuflowlab/FLOWUnsteady #130: Wopwop3 access permission issue

AndreaRapi, 2024-01-08:
I tried again to simulate the aeroacoustic part of a tutorial case (rotorhover) but I find an issues related the **Run PSU-WOPWOP** part of the _FLOWUnsteady_noise_wopwop code_ and the error is the following "IOError: could not spawn `./wopwop3`: permission denied (EACCES)". I already checked and the folder wopwop3 is created as well in the pww path. 
Attached is the code response: [wopwop3_issue.txt](https://github.com/byuflowlab/FLOWUnsteady/files/13864764/wopwop3_issue.txt)

May I ask for your kind help on this issue? 
Andrea

EdoAlvarezR, 2024-01-08:
Hi Andrea, I'm glad to hear you are making progress!

`IOError: could not spawn `./wopwop3`: permission denied (EACCES)` sounds like a system level issue. Meaning, that your user does not have the privileges to read `wopwop3`. Are you able to run the examples that come with PSU-WOPWOP? (standalone executable that is completely independent of FLOWUnsteady)

The following command should grant permissions to everyone to read, write, and execute the binary file:
```bash
chmod 777 path/to/wopwop3
```
See this link to learn more about permissions in Unix systems: [LINK](https://www.linode.com/docs/guides/modify-file-permissions-with-chmod/)

AndreaRapi, 2024-01-09:
Thank you Eduardo.
I did the command to both the folder when I have PSU-WOPWOP files. And I got the following:

- Path to PSU-WOPWOP binary

```
$ ls -l /home/rapi/PSU/PSU-WOPWOP_v3.5.0/Windows
total 23860
-rwxrwxrwx 1 rapi rapi    78336 Jan  4 10:35 checkSuite.exe
-rwxrwxrwx 1 rapi rapi 24348160 Jan  4 10:35 wopwop3.5.0_e36db3b.exe
```

- Path to wopwop3
```
$ ls -l /home/rapi/rotorhover_mid-pww
total 12
-rw-r--r-- 1 rapi rapi   69 Jan  9 09:32 cases.nam
drwxr-xr-x 2 rapi rapi 4096 Jan  9 09:32 runcase
drwxr-xr-x 2 rapi rapi 4096 Jan  9 09:32 wopwop3
```

But the simulation give me the same issue related to the permission. I used the PSU-WOPWOP code separately putting the PSU-WOPWOP executable in the folder and the code is running the examples I guess, because it ends without info.
[image]
If I run the checkSuite I got this issue too.
[image]

Do you have any suggestion?

Many thanks in advance,
Andrea

EdoAlvarezR, 2024-01-09:
I imagine that the zip file that you got with the binary came with some example cases that you can run without needing FLOWUnsteady. Could you try running those examples? If the examples run without issues, then try replacing the example case input files with the case files that FLOWUnsteady generated. Please let me know if that works.

AndreaRapi, 2024-01-10:
I simulated the examples in the zip file and I got the correct output files (OASPLdB, OASPLdBA, pressure and spl_spectrum). I did the same with my FLOWUnsteady case (moving the cases.nam and the runcase folder) and and I get the one shown in the photo. 
[image]

The code appears to be blocked or something...I waited an hour before closing it without getting the output files. 
Then I tried many times, as administrator too, and then I got this
[image]

I checked the runcase folder and I think that I got all the correct outputs that then FLOWUnsteady use to do the postprocessing after the BPM part.
[image]

What do you think about it? I still don't understand why it is not working on julia.

EdoAlvarezR, 2024-01-10:
Thanks for checking that!

The only thing I can think of is that perhaps PSU-WOPWOP v3.5.0 has breaking changes in what it expects to find in the input files. Any chance you could try emailing Prof. Brentner to get PSU-WOPWOP v3.4.4 and see if you run into the same issue?

AndreaRapi, 2024-01-18:
I asked but I am still waiting for the response. Currently, I am utilizing version 3.5.0, and I attempted a sensitivity analysis on certain parameters to scrutinize the code. Changing only the number of revolutions I obtain these two strange results 

[image]

[image]

Do you think that the PSU-WOPWOP code is working badly?

AndreaRapi, 2024-01-25:
Hi @EdoAlvarezR 

I did again the simulation with Linux version and it is work. However, the problem related to accessing the wopwop3 folder remains. Would there be a way to modify your code on this point?
Many thanks in advance.

EdoAlvarezR, 2024-01-25:
Did you try using the chmod command in my previous comment? Make sure that your user has permission to execute the file. You can also do that right-clicking the file and checking the properties (if you are in Linux with a graphic interface)

AndreaRapi, 2024-01-25:
Yes I used the chmod command to improve the permission but still the code can not access to the folder but If I am doing this 
```
$ chmod +x wopwop3_linux_v3.5.0_e36db3b
$ ./wopwop3_linux_v3.5.0_e36db3b
```
in ubuntu the code is working and it is producing the correct output files. I will try to check the user permission again.

EdoAlvarezR, 2024-01-25:
That's is exactly what `run_noise_wopwop(...)` should already be doing: [LINK](https://github.com/byuflowlab/FLOWUnsteady/blob/master/src/FLOWUnsteady_noise_wopwop.jl#L486-L488)

Not ideal, but you can just let `run_noise_wopwop(...)` crash with the `permission denied` error, then go into the folder that `run_noise_wopwop(...)` created and run `wopwop3` manually.

AndreaRapi, 2024-01-25:
You're right and thanks for the suggestions.

closed 2024-01-25 by AndreaRapi

VE111, 2024-12-17:
Hello！
Could you please send me a copy of the WOPWOP code? I would also like to calculate aerodynamic noise using this.
Thanks！