byuflowlab/FLOWUnsteady #131: Acoustic Post Processing Error for specific simulations

philippmandl, 2024-01-16:
Dear All, 

I am currently running some post processing studies including acoustic analysis. For specific simulations the acoustic analysis runs without any problems within FLOWUnsteady and PSU-WopWop (e.g. for 5730rpm with 5deg time step size), but for some of the simulations (5270rpm with 5deg time step size) I am always running into the following problem: 

_WARNING: No number of observer containers specified.
          Defaulting nbObserverContainers in the Environment in 
          namelist to 1
 WARNING: Recording pressure or spectrum over grid.
          This may be a VERY large file.
 
 ERROR: The timestep in the surface file does not match that in the
        loading file_

Do you have any idea what might be the problem? 

Thanks in advance!

philippmandl, 2024-01-18:
If I slightly adjust the RPM in the "Simulation Parameters" which are used as a reference value from 5270rpm to 5269.99 the post-processing works, but interestingly the following discrepancies are shown when comparing the tonal noise: 

[image]
[image]

Any ideas why this is happening?
Thank you very much!

AndreaRapi, 2024-01-18:
Hi there,
what operating system do you have?
Andrea

philippmandl, 2024-01-18:
Hi, 
I am using Linux. 
Best

AndreaRapi, 2024-01-18:
Okay, thank you. I am using Windows as well. Have you encountered a response similar to this when running WOPWOP (v3.5)? It seems to be stuck in a loop.
[image]

philippmandl, 2024-01-18:
Hi Andrea, 

thanks for the reply. I am currently using WOPWOP v3.5 and didn't encounter a problem for the other RPM; just for the specific one (5270). 

Best

AndreaRapi, 2024-01-18:
Ok, I changed RPM to 6000 RPM and I get strange numbers too.

Thank you,
Andrea

AndreaRapi, 2024-01-19:
I tried with the PSU examples and some FLOWUnsteady tutorials. I wil try with linux version.

philippmandl, 2024-01-22:
Okay thank you - looking forward to your results!

EdoAlvarezR, 2024-01-22:
@christianhauschel in #132 seems to have found what the issue is. Unfortunately I wont have time to debug this anytime soon, so I'd recommend either contacting the group at PSU developing PSU-WOPWOP asking what breaking changes they introduced in the geometry, or request for an older version (v3.4).

philippmandl, 2024-01-23:
I see - thanks for your feedback. Hope to have access to version 3.4.4 soon.

AndreaRapi, 2024-01-25:
Hi @philippmandl,
I did again the simulations with the Linux version and it work. I think that the problem is related to the Windows version. I changed RPM too and the results are fine. 
I contacted them too for the v3.4. Let's keep in touch on it.

philippmandl, 2024-01-25:
Hi @AndreaRapi, many thanks for the information. Interestingly I didn't figure out so far what kind of event happens in the plots I have shared in the beginning of this conversation when I post process the SPL levels for the 1st BPF. But good to hear that the Linux-Version should be correct. BR

AndreaRapi, 2024-01-25:
@philippmandl I will try to simulate the case with 5270 RPM to check it!

philippmandl, 2024-01-25:
@AndreaRapi  thank you very much! Looking forward to your results!

AndreaRapi, 2024-01-27:
@philippmandl which is the case you did? Is it the tutorial one but at 5270 RPM?

philippmandl, 2024-01-29:
hi @AndreaRapi, I have used my own example & airfoil geometry, but please try it with the tutorial with 5270rpm.

AndreaRapi, 2024-02-16:
Hi @philippmandl, I tried to do the example with 5270rpm. FLOWUnsteady works well but the wopwop code did not produce the output file killing very fast the code.

```
********** PSU-WOPWOP Version 3.5.0 (Build e36db3b) ***********
 Opening: cases.nam

 *****************************************************************
 ***  Reading case file: ./runcase/runcase.nam
 *****************************************************************
```
When I changed the rpm value to 5270.1 or 5269.9 it is works.
[image]
[image]

Let me know your thoughts. If you need I can provide all the codes, results, etc...

Ciao,
Andrea

EdoAlvarezR, 2024-02-19:
That's really odd. I have run into similar issues with the number of blade elements (some number of blade elements make PSU-WOPWOP crash without a reasonable explanation). Unfortunately, there isn't much we can do to debug and pin point what the issue is since PSU-WOPWOP is a closed-source code 😞

philippmandl, 2024-02-20:
Hi @AndreaRapi, I had similar issues --> changing the value to 5269.99 made a difference in the stability of PSU-WOPWOP, but the results where not realistic ...

AndreaRapi, 2024-02-20:
Hi @philippmandl, maybe we can inform PSU-WOPWOP code developers about this issue...