byuflowlab/FLOWUnsteady #139: Error in compiling ExaFMM running the script build.sh

NM1293, 2024-02-05:
Ello everyone.

I am trying to install the FLOWUnsteady tool following these instructions: https://flow.byu.edu/FLOWUnsteady/installation/general/

However, when I try to compile the ExaFMM using the script build.sh, I receive this error:

mpicxx -DHAVE_CONFIG_H -DJULIA_ENABLE_THREADING -Dhello_EXPORTS -IC:\\Users\\nicol\\.julia\\artifacts\\ebc89b7d2f778b5ffcea561b490b36e7d5f06fa9/include -IC:\\Users\\nicol\\AppData\\Local\\Programs\\Julia-1.8.5\\include/julia -Wunused-parameter -Wextra -Wreorder -std=gnu++1z -O3 -DNDEBUG -fPIC -ffast-math  -I. -I..  -DEXAFMM_WITH_OPENMP  -msse3 -mavx -mavx2 -mavx512f -mavx512cd -mavx512vl -mavx512bw -mavx512dq -mavx512ifma -mavx512vbmi -DNDEBUG -DEXAFMM_EAGER  -funroll-loops -fabi-version=6 -Wfatal-errors -fopenmp  -g -O2  -MT fmm-fmm.o -MD -MP -MF .deps/fmm-fmm.Tpo -c -o fmm-fmm.o `test -f 'fmm.cxx' || echo './'`fmm.cxx
fmm.cxx:10:11: fatal error: jlcxx/jlcxx.hpp: No such file or directory
   10 |  #include "jlcxx/jlcxx.hpp"  //C++ wrapper for julia
      |           ^~~~~~~~~~~~~~~~~

and the required file fmm.so under src/ is not generated.

What could be the error?

Thank you.

cibinjoseph, 2024-02-05:
It looks like the link to your Julia CxxWrap library file is incorrect or the file is missing in that location.

I would run `import CxxWrap; CxxWrap.prefix_path()` in Julia and check the location that it prints out to see if `jlcxx/jlcxx.hpp` is present at that location. 

It should print out that artifacts folder you have in the error message (`C:\Users\nicol\.julia\artifacts\ebc89b7d2f778b5ffcea561b490b36e7d5f06fa9/include`)

If everything seems to look fine, I would remove and re-install CxxWrap and see if that works.

Also which version of Julia are you using?

NM1293, 2024-02-19:
Hello @cibinjoseph, and thank you for your help.
However, it turns out the issue was different, and I managed to resolve it. I made some errors in the initial part of the tutorial, particularly in the installation of Julia. In case it might help someone else, I discovered the correct procedure by first setting up WSL, Julia, and Paraview following these instructions: https://flow.byu.edu/FLOWUnsteady/installation/windows/#windows

and then proceeding with the steps outlined here: https://flow.byu.edu/FLOWUnsteady/installation/general/

from the PyCall installation onwards.

I hope this could be useful for someone.

closed 2024-02-19 by NM1293