byuflowlab/FLOWUnsteady #140: Variable Fidelity - High fidelity tutorial

AndreaRapi, 2024-02-15:
I tried to simulate the High Fidelity DJI 9443 Rotorhover tutorial considering the following:

```
vpm_SFS       = vpm.DynamicSFS(vpm.Estr_fmm, vpm.pseudo3level_positive;
                                   alpha=0.999, maxC=1.0,
                                   clippings=[vpm.clipping_backscatter])
```

but after some iterations (800) Julia killed the process. Is there anything that can be done to fix the problem or understand why?

Thank you,
Andrea

EdoAlvarezR, 2024-02-19:
The only thing I can think off is that perhaps you ran out of RAM and swap memory and your operative system killed the process because of this. You can check that by re-running the simulation while monitoring the RAM usage in your operative system (e.g., `htop` in Linux/MacOS, Task Manager in Windows)

AndreaRapi, 2024-02-20:
Many thanks. I will do it.

closed 2024-02-20 by AndreaRapi

NM1293, 2024-04-24:
@AndreaRapi Hello Andrea.
Have you solved the problem?
Have you understood what it depends on?
I am facing the same problem with the high fidelity case, at iteration 1200.

Thank you.

AndreaRapi, 2024-04-24:
@NM1293 Hi there,
I haven't run that case again because as already suggested it's related to the RAM limit, so I guess is related to the features of my PC. I don't know how to solve it honestly.

NM1293, 2024-04-24:
@AndreaRapi Ok. Thank you for your reply.