byuflowlab/FLOWUnsteady #161: Segmentation fault:

P-Nishanth, 2024-04-09:
Hey, every time I run an example apart from the simple wing I get the following error:

signal (11): Segmentation fault
in expression starting at /home/nish/.julia/packages/FLOWUnsteady/MpOG2/examples/vahana/vahana.jl:376
unknown function (ip: 0x7fffed0735da)
Allocations: 6767744745 (Pool: 6767383817; Big: 360928); GC: 179
Segmentation fault (core dumped)

Sometimes, the system even crashes. 

I have dual-booted Windows and Ubuntu 22 and I'm running flowunsteady on Ubuntu. I have a 24-core machine.

Is this a hardware issue? How do I fix this?
Thanks.

cibinjoseph, 2024-04-09:
Segfaults are really hard to debug. I wonder if you're running out of memory.
Try reducing the discretization on the wings and props and reducing the number of timesteps. See if the segfault exists for a really coarse case.

P-Nishanth, 2024-04-10:
[image]
This is a snip of htop running parallel to FLOWunsteady and the memory consumption is super low.

cibinjoseph, 2024-04-10:
You're workstation's specs should be fine for these cases to run fine.
I would check in htop if the memory goes above the limits when it's close to throwing the segfault.

Since the simple wing case works fine, I would try running it for a longer time and see if it throws a segfault when there's close to the number of particles in this simulation. Or simplify this case and see if it runs for longer timesteps. I'm still inclined to think some sort of memory limit is being hit by these simulations. Not sure why/how though.
I've had issues with memory limits imposed by Linux on the stack size sometimes. The command `ulimit -s unlimited` resolves those. Again, I'm not sure if that's the cause here.

We're in the process of phasing out the FMM package (FLOWExaFMM) for a native Julia one because it has given us a lot of trouble. So hopefully this issue will also be resolved with that switch.

EdoAlvarezR, 2024-04-10:
Really odd. The stack trace points out that the error is on the C++ part of FLOWExaFMM. As @cibinjoseph mentions, my guess is that there is an issue with the amount of memory that the process is allow to use.

P-Nishanth, 2024-04-11:
(@v1.8) pkg> test CxxWrap
     Testing CxxWrap
      Status `/tmp/jl_M8GCHQ/Project.toml`
  [6e4b80f9] BenchmarkTools v1.5.0
⌃ [1f15a43c] CxxWrap v0.11.2
  [1914dd2f] MacroTools v0.5.13
⌅ [3eaa8342] libcxxwrap_julia_jll v0.8.8+1
  [8f399da3] Libdl `@stdlib/Libdl`
  [8dfed614] Test `@stdlib/Test`
      Status `/tmp/jl_M8GCHQ/Manifest.toml`
  [6e4b80f9] BenchmarkTools v1.5.0
⌃ [1f15a43c] CxxWrap v0.11.2
  [692b3bcd] JLLWrappers v1.5.0
  [682c06a0] JSON v0.21.4
  [1914dd2f] MacroTools v0.5.13
  [69de0a69] Parsers v2.8.1
  [aea7be01] PrecompileTools v1.2.1
  [21216c6a] Preferences v1.4.3
⌅ [3eaa8342] libcxxwrap_julia_jll v0.8.8+1
  [0dad84c5] ArgTools v1.1.1 `@stdlib/ArgTools`
  [56f22d72] Artifacts `@stdlib/Artifacts`
  [2a0f44e3] Base64 `@stdlib/Base64`
  [ade2ca70] Dates `@stdlib/Dates`
  [f43a241f] Downloads v1.6.0 `@stdlib/Downloads`
  [7b1f6079] FileWatching `@stdlib/FileWatching`
  [b77e0a4c] InteractiveUtils `@stdlib/InteractiveUtils`
  [b27032c2] LibCURL v0.6.3 `@stdlib/LibCURL`
  [76f85450] LibGit2 `@stdlib/LibGit2`
  [8f399da3] Libdl `@stdlib/Libdl`
  [37e2e46d] LinearAlgebra `@stdlib/LinearAlgebra`
  [56ddb016] Logging `@stdlib/Logging`
  [d6f4376e] Markdown `@stdlib/Markdown`
  [a63ad114] Mmap `@stdlib/Mmap`
  [ca575930] NetworkOptions v1.2.0 `@stdlib/NetworkOptions`
  [44cfe95a] Pkg v1.8.0 `@stdlib/Pkg`
  [de0858da] Printf `@stdlib/Printf`
  [9abbd945] Profile `@stdlib/Profile`
  [3fa0cd96] REPL `@stdlib/REPL`
  [9a3f8284] Random `@stdlib/Random`
  [ea8e919c] SHA v0.7.0 `@stdlib/SHA`
  [9e88b42a] Serialization `@stdlib/Serialization`
  [6462fe0b] Sockets `@stdlib/Sockets`
  [2f01184e] SparseArrays `@stdlib/SparseArrays`
  [10745b16] Statistics `@stdlib/Statistics`
  [fa267f1f] TOML v1.0.0 `@stdlib/TOML`
  [a4e569a6] Tar v1.10.1 `@stdlib/Tar`
  [8dfed614] Test `@stdlib/Test`
  [cf7118a7] UUIDs `@stdlib/UUIDs`
  [4ec0a83e] Unicode `@stdlib/Unicode`
  [e66e0078] CompilerSupportLibraries_jll v1.0.1+0 `@stdlib/CompilerSupportLibraries_jll`
  [deac9b47] LibCURL_jll v7.84.0+0 `@stdlib/LibCURL_jll`
  [29816b5a] LibSSH2_jll v1.10.2+0 `@stdlib/LibSSH2_jll`
  [c8ffd9c3] MbedTLS_jll v2.28.0+0 `@stdlib/MbedTLS_jll`
  [14a3606d] MozillaCACerts_jll v2022.2.1 `@stdlib/MozillaCACerts_jll`
  [4536629a] OpenBLAS_jll v0.3.20+0 `@stdlib/OpenBLAS_jll`
  [83775a58] Zlib_jll v1.2.12+3 `@stdlib/Zlib_jll`
  [8e850b90] libblastrampoline_jll v5.1.1+0 `@stdlib/libblastrampoline_jll`
  [8e850ede] nghttp2_jll v1.48.0+0 `@stdlib/nghttp2_jll`
  [3f19e933] p7zip_jll v17.4.0+0 `@stdlib/p7zip_jll`
        Info Packages marked with ⌃ and ⌅ have new versions available, but those with ⌅ are restricted by compatibility constraints from upgrading.
     Testing Running tests...
Running tests from basic_types.jl...
char               -> CxxChar
signed char        -> Int8
unsigned char      -> UInt8
short              -> Int16
unsigned short     -> UInt16
int                -> Int32
unsigned int       -> UInt32
long               -> Int64
[... 59 lines cut ...]
   [13] eval
      @ ./boot.jl:368 [inlined]
   [14] exec_options(opts::Base.JLOptions)
      @ Base ./client.jl:276
   [15] _start()
      @ Base ./client.jl:522
  in expression starting at /home/nish/.julia/packages/CxxWrap/OcN1Z/test/functions.jl:17
Test Summary:     | Pass  Error  Total  Time
CxxWrap tests     |   98      1     99  5.6s
  basic_types     |   71            71  0.5s
  containers      |   16            16  0.8s
  cxxdereference  |    4             4  0.1s
  except          |    6             6  0.0s
  extended_module |    1             1  0.0s
ERROR: LoadError: Some tests did not pass: 98 passed, 0 failed, 1 errored, 0 broken.
in expression starting at /home/nish/.julia/packages/CxxWrap/OcN1Z/test/runtests.jl:15
ERROR: Package CxxWrap errored during testing

@cibinjoseph The memory consumption does not shoot up at any point.
@EdoAlvarezR Since you indicated that the error might be in the C++ part I removed the package CxxWrap and added it back. Now I get this error. 
Do you guys know how do I to fix this?

Thanks.

PS: Strange that I'm still able to run simulations with this error.

ChrisPsenica, 2026-03-11:
@P-Nishanth I do not have an I9, my workstation has a threadripper 3960x which is 24 core. The issue is not consistent though. Sometimes I get this error, sometimes I do not. This confuses me a bit more as it is harder to find the cause.