byuflowlab/FLOWUnsteady #171: Error while generating geometry

Ajaykrishna0703, 2024-05-07:
Hi all
New user to FlowUnsteady, I am trying to simulate my own propeller in FlowUnsteady, I have generated all the required files for the simulation like propeller distributions and airfoils and i have added those files in rotor folder. While running the file i am getting a error which says "ERROR: LoadError: BoundsError: attempt to access 8×2 DataFrame at index [1, 3]"

cibinjoseph, 2024-05-07:
We'd need more information to figure this out. Can you post the entire stacktrace that gets printed out during the error?

Ajaykrishna0703, 2024-05-07:
> We'd need more information to figure this out. Can you post the entire stacktrace that gets printed out during the error?

this is the above error i am getting
[image]

cibinjoseph, 2024-05-07:
Looks like one of your input propeller geometry files might have an issue. Verify that the format is the same as in the example files.
If that doesn't work, try swapping out each distribution file (like the twistdist, pitchdist etc.) and re-running it, to see when the error pops up.

Ajaykrishna0703, 2024-05-08:
> Looks like one of your input propeller geometry files might have an issue. Verify that the format is the same as in the example files. If that doesn't work, try swapping out each distribution file (like the twistdist, pitchdist etc.) and re-running it, to see when the error pops up.

you mean like swapping out each distribution file with the default files that are already available in database folder and find out which is causing the problem

cibinjoseph, 2024-05-08:
Yep

Ajaykrishna0703, 2024-05-08:
> Yep

Thanks il try it and get back with you

sakurakikiki, 2024-08-21:
Hello, may I ask why the section starts from 0 when entering the propeller geometry file?