byuflowlab/FLOWUnsteady #174: Permanently occupied space after simulation

NM1293, 2024-05-16:
Hello Everyone,
I am utilizing FLOWUnsteady on a second PC, distinct from my usual first PC. On this second PC, it appears that the simulation is consuming storage space, yet even after I delete the simulation results, this space remains occupied.

Contrary to this, such behavior does not occur on my first PC. In fact, during the simulation execution on the first PC, I do not observe any increase in the storage utilization on the C drive until I transfer the simulation results post-completion.

Has this happened to anyone?

What could be the cause of this behavior on the second PC? What is consuming the memory, and how can I reclaim it?

Checking the storage through the PC settings (second PC), I noticed that the hibernation files amount to a total of 204 GB. Could this be related to running the simulation with FLOWUnsteady?

Thank You.

surajrkashyap, 2024-05-23:
Are you using WSL2? Just pitching in since I had a similar issue in the past. Please ignore if this is not the case for you.

WSL2 creates a Virtual Hard Disk (VHD) to store files, and this is automatically resized based on requirements. This storage is not always automatically released even after deleting files within WSL. You may have to compact the VHD to release the storage space. https://learn.microsoft.com/en-us/windows/wsl/disk-space

NM1293, 2024-05-24:
Hello @surajrkashyap,
Thank you for your help. I conducted an investigation into this issue and found that a large amount of space is occupied by a file related to Ubuntu. However, I still do not know if this file is related to the VHD you mentioned.