byuflowlab/FLOWUnsteady #178: Julia Pkd.add error while adding FLOWVLM package

sankprasad, 2024-06-03:
I am installing the FLOWUnsteady software as per the directions in the website. I could run julia and paraview properly. While installing other packages, I am getting the following error. 
I followed the instructions from "https://flow.byu.edu/FLOWUnsteady/installation/general/". Request help ASAP as I am a student and need to do flow simulations for my project. 

I tried to install imp package, but google says that the package is not available for python3
Thanks in advance.

Precompiling project...
  ✗ FLOWVLM
  2 dependencies successfully precompiled in 6 seconds. 172 already precompiled.

ERROR: The following 1 direct dependency failed to precompile:

FLOWVLM [1a3ff0be-0410-4572-aa62-b496bdd1f33b]

Failed to precompile FLOWVLM [1a3ff0be-0410-4572-aa62-b496bdd1f33b] to "/home/pavan/.julia/compiled/v1.10/FLOWVLM/jl_cOPHm0".
ERROR: LoadError: InitError: PyError (PyImport_ImportModule

The Python package imp could not be imported by pyimport. Usually this means
that you did not install imp in the Python version being used by PyCall.

PyCall is currently configured to use the Python version at:

/usr/bin/python3

and you should use whatever mechanism you usually use (apt-get, pip, conda,
etcetera) to install the Python package containing the imp module.

One alternative is to re-configure PyCall to use a different Python
version on your system: set ENV["PYTHON"] to the path/name of the python
executable you want to use, run Pkg.build("PyCall"), and re-launch Julia.

Another alternative is to configure PyCall to use a Julia-specific Python
distribution via the Conda.jl package (which installs a private Anaconda
Python distribution), which has the advantage that packages can be installed
and kept up-to-date via Julia.  As explained in the PyCall documentation,
set ENV["PYTHON"]="", run Pkg.build("PyCall"), and re-launch Julia. Then,
To install the imp module, you can use `pyimport_conda("imp", PKG)`,
where PKG is the Anaconda package that contains the module imp,
or alternatively you can use the Conda package directly (via
`using Conda` followed by `Conda.add` etcetera).

sakurakikiki, 2024-06-22:
Dude, is this problem solved？

sankprasad, 2024-06-24:
I installed a lower version of the python (3.8) and now there is no error. Thanks for your concern

closed 2024-06-24 by sankprasad