byuflowlab/FLOWUnsteady #179: Issues importing FLOWUnsteady package receiving a NULL PyObject error

mortimer-pc, 2024-06-05:
I have followed the installation instructions and have not had any issues until trying to import the FLOWUnsteady package to run some of the example cases. I am receiving a NULL PyObject LoadError during the precompilation.  Please see attached screenshot with details on the error. 

Please advise on a solution. 

[image]

Kalley2, 2024-06-19:
I'm experiencing the same problem, may I ask how you solved it?

mortimer-pc, 2024-06-19:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

Unfortunately, I have not solved it yet.

The information contained in this e-mail and any attachments from Piasecki Aircraft Corporation may contain confidential and/or proprietary information, and is intended only for the named recipient to whom it was originally addressed. If you are not the intended recipient, any disclosure, distribution, or copying of this e-mail or its attachments is strictly prohibited. If you have received this e-mail in error, please notify the sender immediately by return e-mail and permanently delete the e-mail and any attachments.

cibinjoseph, 2024-07-01:
Adding a workaround for this. This is caused due to some weird incompatibility with PyPlot.
Commenting out [everything](https://github.com/byuflowlab/FLOWUnsteady/blob/53e94a4a711239cabda0ede7d57e31a294f8a47c/src/FLOWUnsteady_utils.jl#L29) inside the `formatpyplot()` function in `src/FLOWUnsteady_utils.jl` should allow you to install it. This is only a hotfix for now until we figure out what exactly is going wrong.

sakurakikiki, 2024-07-08:
> Adding a workaround for this. This is caused due to some weird incompatibility with PyPlot. Commenting out [...]

Hello, why do I still have the following problem according to what you said
[image]

EdoAlvarezR, 2024-07-20:
It's a little embarrassing that it took me so long to fix this (it literally took 30 seconds to push a fix), but, in my defense, I just had a baby ¯\\_(ツ)\_/¯.

As @cibinjoseph pointed out, `formatpyplot()` is the cause of the issue in the newer version of Julia (v1.10+), so [I commented out](https://github.com/byuflowlab/FLOWUnsteady/commit/e99e5e44144c0cb9f5621833a6a6c8adddb800e4) the call to `formatpyplot()` upon importing FLOWUnsteady. I had fixed that in my current feature dev branch, but it hadn't make its way to dev and master yet.

closed 2024-07-20 by EdoAlvarezR