byuflowlab/FLOWUnsteady #185: Mach Correction Error During Rotor Hover Simulation at 6000 RPM

AndreaRapi, 2024-07-16:
While simulating a rotor hover at 6000 RPM, I encountered the following error after 570 out of 720 steps:

`ERROR:` LoadError: Mach correction requested on Ma = 1.898085013254046 >= 1.0
[LoadError Mach correction req.txt](https://github.com/user-attachments/files/16247397/LoadError.Mach.correction.req.txt)

I attempted to examine the functions to identify the specific correction causing this issue but could not find anything related.
Could you please help me resolve this error?

Thank you,
Andrea

EdoAlvarezR, 2024-07-20:
Hi Andrea! 

The error is exactly what it says: The flow became supersonic somewhere along the blade. The error is thrown because the blade elements use a Prandtl–Glauert compressibility correction that is valid up to a local Mach inflow of about 0.6. 

The error is intentionally thrown to kill the simulation because it typically means that your simulation has blown up, so I recommend visualizing the simulation to determine why it blew up.

EdoAlvarezR, 2024-07-20:
You can also confirm that the simulation has blown up by taking a look at the enstrophy monitor.