byuflowlab/FLOWUnsteady #229: FLOWUnsteady precompilation fails with `UndefVarError: AbstractWing not defined`

patterson117, 2025-08-15:
**Description:** Importing FLOWUnsteady fails (`import FLOWUnsteady as uns`) in `FLOWUnsteady_processing_force.jl:84` where `AbstractWing` type is referenced but not defined/imported.

**Environment:**
Windows 11
WSL Ubuntu 22.04

**Steps to Reproduce:** Occurs when following the [tutorial](https://flow.byu.edu/FLOWUnsteady/installation/general/#Test-FLOWUnsteady), all prior mandatory steps completed and tests passed.

**Pkg Status**
```julia
(@v1.10) pkg> status
Status `~/.julia/environments/v1.10/Project.toml`
  [008c61af] AirfoilPrep v2.1.2 `https://github.com/byuflowlab/AirfoilPrep.jl#v2.1.2`
  [f91b385c] BPM v2.0.1 `https://github.com/byuflowlab/BPM.jl#v2.0.1`
⌃ [1f15a43c] CxxWrap v0.15.0
  [a07d1f4e] FLOWExaFMM v2.1.1 `~/flowunsteady/FLOWExaFMM`
  [d27480ee] FLOWNoise v2.3.3 `https://github.com/byuflowlab/FLOWNoise#v2.3.3`
  [b491798d] FLOWUnsteady v3.4.0 `https://github.com/byuflowlab/FLOWUnsteady#master`
  [1a3ff0be] FLOWVLM v2.1.3 `https://github.com/byuflowlab/FLOWVLM#v2.1.3`
  [6e19019d] FLOWVPM v4.0.0 `https://github.com/byuflowlab/FLOWVPM.jl#master`
  [83792f5e] GeometricTools v2.3.3
  [438e738f] PyCall v1.96.4
  [9b3f6a95] VSPGeom v0.6.6
```
**Detailed Error**
```julia
julia> import FLOWUnsteady as uns
Precompiling FLOWUnsteady
        Info Given FLOWUnsteady was explicitly requested, output will be shown live
ERROR: LoadError: UndefVarError: `AbstractWing` not defined
Stacktrace:
 [1] getproperty(x::Module, f::Symbol)
   @ Base ./Base.jl:31
 [2] top-level scope
   @ ~/.julia/packages/FLOWUnsteady/L8fkM/src/FLOWUnsteady_processing_force.jl:84
 [3] include(mod::Module, _path::String)
   @ Base ./Base.jl:495
 [4] include
   @ ~/.julia/packages/FLOWUnsteady/L8fkM/src/FLOWUnsteady.jl:11 [inlined]
 [5] top-level scope
   @ ~/.julia/packages/FLOWUnsteady/L8fkM/src/FLOWUnsteady.jl:65
 [6] include
   @ ./Base.jl:495 [inlined]
 [7] include_package_for_output(pkg::Base.PkgId, input::String, depot_path::Vector{String}, dl_load_path::Vector{String}, load_path::Vector{String}, concrete_deps::Vector{Pair{Base.PkgId, UInt128}}, source::Nothing)
   @ Base ./loading.jl:2285
 [8] top-level scope
   @ stdin:3
in expression starting at ~/.julia/packages/FLOWUnsteady/L8fkM/src/FLOWUnsteady_processing_force.jl:84
in expression starting at ~/.julia/packages/FLOWUnsteady/L8fkM/src/FLOWUnsteady.jl:1
in expression starting at stdin:3
  ✗ FLOWUnsteady
  0 dependencies successfully precompiled in 5 seconds. 203 already precompiled.
```

patterson117, 2025-08-15:
**Possible fix:** Pulled explicit revision numbers out of install of byuflowlab github packages, so that all repositories were on latest branch in [Tutorial>Other-Packages](https://flow.byu.edu/FLOWUnsteady/installation/general/#Other-Packages).

```julia
(@v1.11) pkg>status
Status `~/.julia/environments/v1.11/Project.toml`
  [008c61af] AirfoilPrep v2.1.3 `https://github.com/byuflowlab/AirfoilPrep.jl#master`
  [f91b385c] BPM v3.0.0 `https://github.com/byuflowlab/BPM.jl#master`
⌃ [1f15a43c] CxxWrap v0.15.0
  [a07d1f4e] FLOWExaFMM v2.1.1 `../../../Programs/FLOWExaFMM`
  [d27480ee] FLOWNoise v2.3.3 `https://github.com/byuflowlab/FLOWNoise#master`
  [b491798d] FLOWUnsteady v3.4.0 `https://github.com/byuflowlab/FLOWUnsteady#master`
  [1a3ff0be] FLOWVLM v2.1.4 `https://github.com/byuflowlab/FLOWVLM#master`
  [6e19019d] FLOWVPM v4.0.0 `https://github.com/byuflowlab/FLOWVPM.jl#master`
  [83792f5e] GeometricTools v2.3.3
  [438e738f] PyCall v1.96.4
  [d330b81b] PyPlot v2.11.6
  [9b3f6a95] VSPGeom v0.6.6
```

Kim-Yunsu, 2025-08-19:
Thank you for your reply. :)

kaichenkai, 2025-08-24:
[image]

This picture is clearer to see

EdoAlvarezR, 2025-08-25:
@BTV25 @rymanderson @atagg2 this is related to breaking changes introduced into main. Please update tutorials and API as needed.

patterson117, 2025-08-27:
I started with a clean setup and walked through the tutorial. It looks like the documentation is not yet updated but I replaced the [FLOWVPM](https://flow.byu.edu/FLOWUnsteady/installation/general/#FLOWVPM) (after installing CxxWrap), [FLOWVPM](https://flow.byu.edu/FLOWUnsteady/installation/general/#Other-Packages), and [Add-FLOWUnsteady](https://flow.byu.edu/FLOWUnsteady/installation/general/#Add-FLOWUnsteady) sections with the [solution above](https://github.com/byuflowlab/FLOWUnsteady/issues/229#issuecomment-3229023415).

I was able to successfully run the simple wing example.

closed 2025-08-27 by patterson117