byuflowlab/FLOWUnsteady #230: loft.vtk is not created.

Kim-Yunsu, 2025-08-19:
Hello. My environment is as follows.
- Windows 11
- WSL2 Ubuntu-22.04
- Paraview: 5.9.1
- Julia: 1.10.7

I completed the environment configuration following the attached ‘FLOWUnsteady Installation_250819.txt’ and ran ‘rotorhover.jl’.
In general, loft.vtk should be generated.  
However, I could only find 'vlm.vtk' and 'compact.vtk'.

I have attached the execution terminal log.

If you have experienced the same issue or have resolved it, please let me know how to resolve it.

[FLOWUnsteady Installation_250819.txt](https://github.com/user-attachments/files/21851690/FLOWUnsteady.Installation_250819.txt)

[terminal.log](https://github.com/user-attachments/files/21851836/terminal.log)

BTV25, 2025-08-27:
@Kim-Yunsu, intall instructions have been modified, please reinstall and test again. I am getting loft files with my tests.

Kim-Yunsu, 2025-08-29:
@BTV25 
Now I can obtain the loft file. Thank you.
However, I have one question. I understand that particle-particle velocity calculations are performed using ExaFMM. Could you explain why it is no longer needed?

BTV25, 2025-08-29:
@Kim-Yunsu, that is a great question. We no longer need ExaFMM because we have developed a pure Julia implementation of Fast Multipole Method in [FastMultiopole.jl](https://github.com/byuflowlab/FastMultipole.jl) to do those calculations.

Kim-Yunsu, 2025-09-01:
@BTV25 
I have been using a command to enable multi-threading when running Julia. The reason I previously reported an issue where multi-threading wasn't working is because when I run the analysis using the command to enable multi-threading, I get the following error. I have attached my log file.
I would like to ask two questions regarding this.

1. When performing multi-threading via the Julia command, a software issue occurs. Do you know of a solution?
2. As with the issue I raised earlier, simply running Julia does not enable parallel execution. Do you know of a solution?

Thank you for always kindly answering my questions.

[20250901_terminal.log](https://github.com/user-attachments/files/22072424/20250901_terminal.log)

BTV25, 2025-09-01:
The error I see is likely a software bug. I am having difficulty replicating the issue. Could you send me the exact script and supporting files you are running?

To run Julia in parallel, I would look at the [guide to multithreading](https://docs.julialang.org/en/v1/manual/multi-threading/) in Julia. As a basic introduction, you have to tell Julia to use threads. You can specify a specific number or just set it to auto, which uses the maximum number of threads. Here are a few examples of launching a Julia script with threads.
`julia --threads=auto script.jl`
`julia -t auto script.jl`
`julia --threads=50 script.jl`
`julia -t 50 script.jl`

Also feel free to use `julia -h` to give the help page, one entry is about threads.

BTV25, 2025-09-02:
@Kim-Yunsu 
The error was found in FastMultipole.jl. A new version is being registered now, v2.0.2. Updating FastMultipole should allow your simulation to run properly.

BTV25, 2025-09-03:
I will try to replicate the issue, but it is likely something to do with your Python environment. I would advise reinstalling Python packages that might try to run in parallel and resetting other threading instructions before running:
export OMP_NUM_THREADS=1
export MKL_NUM_THREADS=1
export OPENBLAS_NUM_THREADS=1